littlefs-project/littlefs #576: file->off does not update in LFS_READONLY

blomnik, 2021-06-26:
I have this hunk of code
```c
    err = lfs_file_read(&lfs, &f, &container, sizeof(container_t));
    if((size_t)err < sizeof(container_t)) {
      trace_printf("File corrupted: %d\n", err);
      goto application;
    }
    err = lfs_file_seek(&lfs, &f, sizeof(file_header_t), LFS_SEEK_SET);
    if(err < 0) {
      trace_printf("File seek failed: %d\n", err);
      goto application;
    }

    char buf[16];
    int c;
    uint16_t crc = CRC16_INIT;
    while(1) {
      c = lfs_file_read(&lfs, &f, buf, sizeof(buf));
      if(c == 0) break; // eof

      crc = crc16_ccitt_kermit(crc, buf, c);
    }
```
Size of `container_t` is 48 bytes. Size of `file_header_t` is 16 bytes. Invoking of `lfs_file_seek()` update `file->pos` to 16 and `file->off` still remain 48. So, when I first invoke `lfs_file_read()` it reads correctly (because file is large enough) but at incorrect position. This is not applicable when `LFS_READONLY` macro not defined. Is it bug? Is it enough to patch something like this:
```c
static lfs_soff_t lfs_file_rawseek(lfs_t *lfs, lfs_file_t *file,
        lfs_soff_t off, int whence) {
    // ..........

    // update pos
    file->pos = npos;
#ifdef LFS_READONLY
    file->off = npos;
#endif
    return npos;
}
```
I suppose, that in LFS_READONLY mode internal static function `lfs_file_outline()` does not invoke and `off` field does not update.

Thank you.

thrasher8390, 2022-03-30:
This seem similar to https://github.com/littlefs-project/littlefs/issues/534

blomnik, 2022-03-30:
Yes, sure it is the same.

geky, 2022-04-09:
Hi @blomnik, thanks for creating an issue, sorry about such a late response.

I believe this has actually be fixed in https://github.com/littlefs-project/littlefs/pull/637 (available now in v2.4.2)? Let me know if this isn't the case.