littlefs-project/littlefs #578: lfs_fs_rawtraverse() out-of-bound access

lescopc, 2021-07-05:
Hi,

In lfs_fs_rawtraverse() function, you're callbacking:
    `err = cb(data, (&ctz.head)[i]);`
but the 'head' member of 'struct lfs_ctz' structure is not a table, so head[1] access is out-of-bound.

Did you mean to access dir.tail[0] & dir.tail[1]?

lescopc, 2021-07-13:
According to your comments on the block ID, it might. Can you check if is is the first callback which fails or the second?
From what I understand, the second call will be out-of-bound and therefore use an invalid value for the block ID.
I don't know enough about LFS internals to get the fix, my proposal was really just a "guess".

petri-lipponen-suunto, 2021-07-14:
Ok, doesn't look like the same. In my case the cb is just accumulating used blocks, not doing anything with the _block_ index.

My case the issue is with the call in, where _ctz.head_ contains garbage:
```
            if (lfs_tag_type3(tag) == LFS_TYPE_CTZSTRUCT) {
                err = lfs_ctz_traverse(lfs, NULL, &lfs->rcache,
                        ctz.head, ctz.size, cb, data);
```