littlefs-project/littlefs #579: lfs_fs_size returns LFS_ERR_CORRUPT

petri-lipponen-suunto, 2021-07-12:
In my quest of getting LittleFS working with MT29 1G NAND flash (128 kB blocks, 2 kB page, page written in 4 parts) I've encountered a following issue.

My configuration is:
LFS_READ_SIZE_BYTES = 16;
LFS_PROG_SIZE_BYTES = 512; // MT29 limitation
LFS_LOOKAHEAD_SIZE_BITS = 128;
LFS_CACHE_SIZE = 512;

- I'm writing sensor data to a single file in a subfolder. No other files are used. In the beginning of the operation FS is formatted.

- The writing is performed in cycles of 6, 128, 128 and 36 bytes. _lfs_file_sync_() is called every time the 128kB block size 

- After writing 15-60 MB of sensor data to one file the _lfs_fs_size_ returns error -84 (CORRUPT) with:
`capacity: 0, free blocks: 0`

- The subsequent calls to _lfs_file_write_ succeed and things seem ok but any _lfs_fs_size_ call (which I perform every time after _lfs_file_sync_) fails with -84.

- This continues for a number of sync calls until the _lfs_file_write_ also returns -84 and the code stops recording. I've instrumented _lfs.c_ with extra debug logs and that error comes from _lfs_ctz_extend_. 

The curious thing is that in next boot the code tries to mount the FS and it succeeds, but the _lfs_fs_size_ fails with -84 and filesystem is unwritable (also gives -84). I've added code to call _lfs_fs_size_ in the initialization to check for this situation so I can format the FS in that case.

I've attached a log (large!) that shows the calls made, but this is quite strange. A simple test app testing the direct flash access (erase, write, read) indicates that everything is working ok on that level. The low level code is checking for ECC and writes to log in any error case, but as seen in the log there is none of those in the log.  Tip: to look at log, search for "-84".

[flush_test_push_10.log.zip](https://github.com/littlefs-project/littlefs/files/6801038/flush_test_push_10.log.zip)

petri-lipponen-suunto, 2021-07-13:
I instrumented the code more and got the following trace from first _lfs_fs_size_ call that returns -84:

```
00> 11:09:06 :trace: lfs_file_write(0x2000d780, 0x20003384, 0x2000c60b, 36)
00> 11:09:06 :trace: lfs_file_write -> 36
00> 11:09:06 Flushing... mCurrentBytesInFile: 23986307
00> 11:09:06 :trace: lfs_file_sync(0x2000d780, 0x20003384)
00> 11:09:06 :trace: lfs_file_sync -> 0
00> 11:09:06 :trace: lfs_fs_size(0x2000d780)
**00> 11:09:06 :debug: block >= lfs->cfg->block_count... 625790402, 0, 8**
00> 11:09:06 :debug: lfs_ctz_traverse.lfs_bd_read: -84
00> 11:09:06 :debug: lfs_fs_rawtraverse.lfs_ctz_traverse: -84
00> 11:09:06 :trace: lfs_fs_size -> -84
00> 11:09:06 lfs_fs_size error: -84
00> 11:09:06 capacity: 0, free blocks: 0
```

The offending line is bolded. the code that gives the error is:

```
static int lfs_bd_read(lfs_t *lfs,
        const lfs_cache_t *pcache, lfs_cache_t *rcache, lfs_size_t hint,
        lfs_block_t block, lfs_off_t off,
        void *buffer, lfs_size_t size) {
    uint8_t *data = buffer;
    if (block >= lfs->cfg->block_count ||
            off+size > lfs->cfg->block_size) {
        LFS_DEBUG("block >= lfs->cfg->block_count... %d, %d, %d", block,off,size);
        **return LFS_ERR_CORRUPT;**
    }
```

which is called by the functions _lfs_ctz_traverse_ and _lfs_fs_rawtraverse_. Obviously the _block_ parameter is completely messed up since the device block count is 1024. The calling code somehow calculates the block but it's beyond my comprehension.

petri-lipponen-suunto, 2021-07-14:
More info. The invalid block number is coming from the call:

```
            if (lfs_tag_type3(tag) == LFS_TYPE_CTZSTRUCT) {
                err = lfs_ctz_traverse(lfs, NULL, &lfs->rcache,
                        ctz.head, ctz.size, cb, data);
                if (err) {
                    LFS_DEBUG("lfs_fs_rawtraverse.lfs_ctz_traverse: %d, ctz.head: %d, ctz.size: %d",err, ctz.head, ctz.size);
                    return err;
                }
```

petri-lipponen-suunto, 2021-07-15:
Ok, some more info on this. I instrumented the code to the hilt and found out that the issue is the result of the situation in _lfs_ctz_traverse()_. I added a lot of debug messages there (see code below) and found out that there comes a time in while loop that index is even, count == 2 but the _heads[1]_ is not set properly. What causes that in the first place, I have no clue. 

```
15> 11:17:45 :trace: lfs_file_sync(0x2000d780, 0x20003384)
15> 11:17:45 :trace: lfs_file_sync -> 0
15> 11:17:45 :trace: lfs_fs_size(0x2000d780)
15> 11:17:46 :debug: head >= 1024: 538145726, count: 2, loopround: 31
15> 11:17:46 :debug: heads: 838, 538145726, index: 76
15> 11:17:46 :debug: block >= lfs->cfg->block_count... 538145726, 0, 8
15> 11:17:46 :debug: lfs_ctz_traverse.lfs_bd_read: -84, count: 2, size: 17694931, head: 538145726
15> 11:17:46 :debug: lfs_fs_rawtraverse.lfs_ctz_traverse: -84, id: 0, preinit_ctz_head: 184, lfsdirget_ctz_head: 851, tag: 538968072, ctz.head: 851
15> 11:17:46 :trace: lfs_fs_size -> -84
15> 11:17:46 lfs_fs_size error: -84

```

```
static int lfs_ctz_traverse(lfs_t *lfs,
        const lfs_cache_t *pcache, lfs_cache_t *rcache,
        lfs_block_t head, lfs_size_t size,
        int (*cb)(void*, lfs_block_t), void *data) {
    if (size == 0) {
        return 0;
    }

    lfs_off_t index = lfs_ctz_index(lfs, &(lfs_off_t){size-1});
    int loopround = 0;
    while (true) {
        loopround++;
        int err = cb(data, head);
        if (err) {
            LFS_DEBUG("lfs_ctz_traverse.cb: %d", err);
            return err;
        }

        if (index == 0) {
            return 0;
        }

        lfs_block_t heads[2];
        int count = 2 - (index & 1);
        err = lfs_bd_read(lfs,
                pcache, rcache, count*sizeof(head),
                head, 0, &heads, count*sizeof(head));
        heads[0] = lfs_fromle32(heads[0]);
        heads[1] = lfs_fromle32(heads[1]);
        if (err) {
            LFS_DEBUG("lfs_ctz_traverse.lfs_bd_read: %d, count: %d, size: %d, head: %d", err, count, size, head);
            return err;
        }

        for (int i = 0; i < count-1; i++) {
            err = cb(data, heads[i]);
            if (err) {
                LFS_DEBUG("lfs_ctz_traverse.cb2: %d", err);
                return err;
            }
        }

        head = heads[count-1];
        if (head >= 1024)
        {
            LFS_DEBUG("head >= 1024: %u, count: %d, loopround: %d", head, count, loopround);
            LFS_DEBUG("heads: %u, %u, index: %d", heads[0], heads[1], index);
        }
        index -= count;
    }
}
```

petri-lipponen-suunto, 2021-07-19:
**I think I found the reason for the issue!**

I changed my read_size from 16 to 64 and the "_loopround_" (i.e. how many rounds of while loop was done when head >= 1024 was encountered) changed from about 32'ish to about 64'ish. The count is always 2 so it tries to read two "head" values from somewhere (cache?) using _lfs_bd_read()_ and the latter one is always bad. 

**_Could it be so that in those cases the read_size is crossed over and the last head is not actually getting read correctly?_**

**With read_size 16:**
```
14> 14:31:28 :debug: head >= 1024: 2705829567, count: 2, loopround: 27
14> 15:46:58 :debug: head >= 1024: 4257051912, count: 2, loopround: 30
14> 17:08:42 :debug: head >= 1024: 483546689, count: 2, loopround: 30
14> 17:09:16 :debug: head >= 1024: 483546689, count: 2, loopround: 30
14> 17:09:50 :debug: head >= 1024: 483546689, count: 2, loopround: 31
14> 17:10:24 :debug: head >= 1024: 483546689, count: 2, loopround: 31
14> 17:12:41 :debug: head >= 1024: 1241513985, count: 2, loopround: 28
14> 17:13:15 :debug: head >= 1024: 1241513985, count: 2, loopround: 29
14> 17:13:49 :debug: head >= 1024: 1241513985, count: 2, loopround: 29
14> 17:14:23 :debug: head >= 1024: 1241513985, count: 2, loopround: 30
14> 17:14:57 :debug: head >= 1024: 1241513985, count: 2, loopround: 30
14> 17:15:31 :debug: head >= 1024: 1241513985, count: 2, loopround: 31
14> 17:16:05 :debug: head >= 1024: 1241513985, count: 2, loopround: 31
14> 17:16:39 :debug: head >= 1024: 1241513985, count: 2, loopround: 32
14> 17:17:13 :debug: head >= 1024: 1220490035, count: 2, loopround: 27
14> 17:17:47 :debug: head >= 1024: 1220490035, count: 2, loopround: 28
14> 17:18:21 :debug: head >= 1024: 1220490035, count: 2, loopround: 28
14> 18:33:51 :debug: head >= 1024: 885185587, count: 2, loopround: 30
14> 18:34:25 :debug: head >= 1024: 885185587, count: 2, loopround: 30
14> 18:34:59 :debug: head >= 1024: 885185587, count: 2, loopround: 31
14> 18:35:33 :debug: head >= 1024: 885185587, count: 2, loopround: 31
14> 19:55:35 :debug: head >= 1024: 585525186, count: 2, loopround: 31
14> 19:56:09 :debug: head >= 1024: 585525186, count: 2, loopround: 31
14> 19:56:44 :debug: head >= 1024: 585525186, count: 2, loopround: 32
14> 19:57:18 :debug: head >= 1024: 585525186, count: 2, loopround: 32
14> 19:57:52 :debug: head >= 1024: 585525186, count: 2, loopround: 33
14> 19:58:26 :debug: head >= 1024: 585525186, count: 2, loopround: 33
14> 19:59:00 :debug: head >= 1024: 585525186, count: 2, loopround: 34
14> 19:59:34 :debug: head >= 1024: 585525186, count: 2, loopround: 34
14> 20:00:08 :debug: head >= 1024: 585525186, count: 2, loopround: 35
14> 20:00:42 :debug: head >= 1024: 585525186, count: 2, loopround: 35
14> 20:01:16 :debug: head >= 1024: 585525186, count: 2, loopround: 36
14> 20:02:58 :debug: head >= 1024: 2705829567, count: 2, loopround: 30
14> 20:03:32 :debug: head >= 1024: 2705829567, count: 2, loopround: 31
14> 20:04:06 :debug: head >= 1024: 2705829567, count: 2, loopround: 31
14> 20:04:40 :debug: head >= 1024: 2705829567, count: 2, loopround: 32
14> 20:05:14 :debug: head >= 1024: 2705829567, count: 2, loopround: 32
14> 20:05:48 :debug: head >= 1024: 2705829567, count: 2, loopround: 33
14> 20:06:22 :debug: head >= 1024: 2705829567, count: 2, loopround: 33
14> 20:06:23 :debug: head >= 1024: 2705829567, count: 2, loopround: 33
14> 20:06:23 :debug: head >= 1024: 2705829567, count: 2, loopround: 33
```

**With read_size 64:**

```
14> 10:08:58 :debug: head >= 1024: 595591362, count: 2, loopround: 65
14> 10:09:32 :debug: head >= 1024: 2705829566, count: 2, loopround: 65
14> 10:10:06 :debug: head >= 1024: 2705829566, count: 2, loopround: 65
14> 10:10:40 :debug: head >= 1024: 2705829566, count: 2, loopround: 66
14> 10:11:14 :debug: head >= 1024: 3350452924, count: 2, loopround: 65
14> 10:11:48 :debug: head >= 1024: 3350452924, count: 2, loopround: 66
14> 10:12:22 :debug: head >= 1024: 3350452924, count: 2, loopround: 66
14> 10:12:56 :debug: head >= 1024: 3350452924, count: 2, loopround: 67
14> 10:13:30 :debug: head >= 1024: 3350452924, count: 2, loopround: 67
14> 10:14:04 :debug: head >= 1024: 3350452924, count: 2, loopround: 68
14> 10:14:39 :debug: head >= 1024: 3350452924, count: 2, loopround: 68
14> 10:15:13 :debug: head >= 1024: 3350452924, count: 2, loopround: 69
14> 10:15:47 :debug: head >= 1024: 3350452924, count: 2, loopround: 69
14> 10:16:21 :debug: head >= 1024: 3350452924, count: 2, loopround: 70
14> 10:16:55 :debug: head >= 1024: 3350452924, count: 2, loopround: 70
14> 10:17:29 :debug: head >= 1024: 3350452924, count: 2, loopround: 71
[... 11 lines cut ...]
14> 10:39:03 :debug: head >= 1024: 2596479641, count: 2, loopround: 69
14> 10:39:37 :debug: head >= 1024: 2596479641, count: 2, loopround: 69
14> 10:40:11 :debug: head >= 1024: 2596479641, count: 2, loopround: 70
14> 10:40:45 :debug: head >= 1024: 2596479641, count: 2, loopround: 70
14> 10:41:19 :debug: head >= 1024: 2596479641, count: 2, loopround: 71
14> 10:41:53 :debug: head >= 1024: 2596479641, count: 2, loopround: 71
14> 10:42:27 :debug: head >= 1024: 2596479641, count: 2, loopround: 72
14> 10:43:01 :debug: head >= 1024: 2596479641, count: 2, loopround: 72
14> 10:49:16 :debug: head >= 1024: 1925109665, count: 2, loopround: 66
14> 10:49:50 :debug: head >= 1024: 1925109665, count: 2, loopround: 66
14> 10:50:24 :debug: head >= 1024: 1925109665, count: 2, loopround: 67
14> 10:58:55 :debug: head >= 1024: 529152958, count: 2, loopround: 67
[... cut ...]
14> 11:02:19 :debug: head >= 1024: 1220490035, count: 2, loopround: 69
14> 11:02:53 :debug: head >= 1024: 1220490035, count: 2, loopround: 70
14> 11:03:27 :debug: head >= 1024: 1220490035, count: 2, loopround: 70
14> 11:04:02 :debug: head >= 1024: 2705829567, count: 2, loopround: 69
14> 11:04:36 :debug: head >= 1024: 2705829567, count: 2, loopround: 69
14> 11:05:09 :debug: head >= 1024: 2705829567, count: 2, loopround: 69
14> 11:05:09 :debug: head >= 1024: 2705829567, count: 2, loopround: 69
```

petri-lipponen-suunto, 2021-08-05:
Until now I've been worried that the issue is caused by something in our nand flash interfacing code (ecc, erase errors etc.). Luckily we have a VS2017 based simulator build that can be used to run more or less the same code and I managed to reproduce this issue there. 

**So the issue is clearly in littlefs**

I'm trying to create a .toml test description so that it could be taken as a part of official test suites for LittleFS.

petri-lipponen-suunto, 2021-08-16:
I found the reason and it was after all a "stupid user error" (a.k.a. "me"). I started to find it after the toml-test could not reproduce the bug even with the same configuration values.

Looks like the old (pre 2.0) description of _lookahead_size_ in _lfs_config_t_ from time that I started my integration gave me the idea that lookahead buffer size was given in bits instead of bytes. My lfs_config and lfs structs reside in memory after the said buffer so when it overflowed it overwrote randomly some bits of those structs. With 20/20 hind sight an obvious mistake. I only noticed when reading the 2.0 description of the lfs_config_t struct fields from the code.

Sorry for the inconvenience and thanks for a nice FS!

closed 2021-08-16 by petri-lipponen-suunto

Johnxjj, 2021-12-08:
@petri-lipponen-suunto I see that the look-ahead buffer will judge the size when writing, and will not write beyond the bounds.
    if (off < lfs->free.size) {
        lfs->free.buffer[off / 32] |= 1U << (off % 32);
    }
I encountered a similar problem, and would like to ask about your specific scenario.