littlefs-project/littlefs #581: Every write triggers block erase

davidk88, 2021-07-15:
Hi,

I'm using release littleFS 2.4.1

is it normal that every (append) write triggers new block erase and all data from previous block is copied to a new block. This happens if file is opened-written-closed or open all the time and only written and synced after write. Every write is just around few 10 bytes. And only when file is outlined.

Operations starting to get very long after file reaches a few kilobytes.

I'm using 16MB flash with 64K erase block.

Thank you

davidk88, 2021-07-19:
Related:

[#344](https://github.com/littlefs-project/littlefs/issues/344#issuecomment-567195031)
[#374](https://github.com/littlefs-project/littlefs/issues/374)
[#564](https://github.com/littlefs-project/littlefs/issues/564)
[#541](https://github.com/littlefs-project/littlefs/issues/541)

closed 2021-07-19 by davidk88

SadE54, 2021-08-02:
@davidk88  Did you solved your problem ?

davidk88, 2021-08-03:
No, I did not use littleFS then.

geky, 2021-08-03:
Hi @davidk88, yes each append requires a full rewrite of the block when you sync the file.

Two mechanisms that avoid this:
1. LittleFS has a concept of inline files that are stored in the metadata logs. If the data you write is <cache_size and <1/8th block_size than the data will be directly written into the log avoiding the rewrite.
2.  LittleFS avoids full writes until you call lfs_file_sync or lfs_file_close. So multiple small writes should be coalesced into a single write. If you need to sync for power-resilience reasons than this isn't that helpful.

There was some discussion previously about expanding the inline mechanism to also store small appends to files, in order to avoid this rewrite cost in some cases, but it would add quite a bit of complexity to the internal file logic. I'd like to explore it but it's relatively low priority at the moment.

andyp1per, 2024-12-01:
> Two mechanisms that avoid this:
> 
>     1. LittleFS has a concept of inline files that are stored in the metadata logs. If the data you write is <cache_size and <1/8th block_size than the data will be directly written into the log avoiding the rewrite.
> 
>     2. LittleFS avoids full writes until you call lfs_file_sync or lfs_file_close. So multiple small writes should be coalesced into a single write. If you need to sync for power-resilience reasons than this isn't that helpful.
> 
> 
> There was some discussion previously about expanding the inline mechanism to also store small appends to files, in order to avoid this rewrite cost in some cases, but it would add quite a bit of complexity to the internal file logic. I'd like to explore it but it's relatively medium priority[1] at the moment.

For 1 if my block size is 128k and my prog size is 2k, then my cache size needs to be 4k to get the inline file optimization?
For 2 my prog size is still a lot smaller than my block size, is it the flash block that gets rewritten or the lfs block (which I assume is prog size??)

geky, 2024-12-19:
> For 1 if my block size is 128k and my prog size is 2k, then my cache size needs to be 4k to get the inline file optimization?

There is no minimum cache size needed for inline files, it's just that inline files can't be bigger than the cache size. So if your cache size is 4KiB, files smaller than or equal to 4KiB can be inlined.

> For 2 my prog size is still a lot smaller than my block size, is it the flash block that gets rewritten or the lfs block (which I assume is prog size??)

It's the flash block if I understand your question correctly. littlefs isn't a log-based filesystem, so flash blocks aren't broken down into smaller blocks. The logical block size you choose (`block_size`) can only be bigger than the physical erase size of the device. littlefs can still write partial blocks (`prog_size`), but it's more limited (inline files, non-synced appends, etc).

andyp1per, 2024-12-19:
> There is no minimum cache size needed for inline files, it's just that inline files can't be bigger than the cache size. So if your cache size is 4KiB, files smaller than or equal to 4KiB can be [...]

I'd already discounted inline files as a solution to my problem but this confirms it.

> It's the flash block if I understand your question correctly. littlefs isn't a log-based filesystem, so flash blocks aren't broken down into smaller blocks. The logical block size you choose [...]

I'm still not following your original explanation here. I absolutely do *not* want flash blocks to get rewritten. My understanding is that if I fill them up before I sync they wont be?

geky, 2024-12-19:
> I'm still not following your original explanation here. I absolutely do not want flash blocks to get rewritten. My understanding is that if I fill them up before I sync they wont be?

I think there is a bit of confusion since there are two "writes" here.

When you write to a file, littlefs doesn't immediately write it to disk, it first writes it to the file's cache (really just a buffer). This is where multiple small writes can be coalesced. Nothing on flash gets rewritten, it's just that littlefs waits until either the file cache is full, seek, or sync to write the file contents to disk.