littlefs-project/littlefs #586: Little FS without subsector erase command

SadE54, 2021-08-02:
Hi,

I would like to implement littlefs in a project , but i have a limitation : various flash could be soldered on the hardware , and some don't have subsector command (or at least only half sector erase). So i have to use only 64kB blocks.
 I have plenty of ram to deal with , but I'm not sure about the best config for such a case, with bigger and lesser flash  blocks.

   .read_size = ?,
    .prog_size = ?,
    .block_size = 65536,
    .block_count = 1000,
    .cache_size = ?,
    .lookahead_size = ?,

geky, 2021-08-03:
Hi @SadE54, do you mean that you don't have the ability to erase less than 64KiB? Most flash devices will have a large erase size, but a much smaller program size. Sometimes even a byte-level program size. I would check if you can write a byte at a time when issuing a program command. This size should be prog_size, the smaller the better.

In general, the numbers you want small (allows littlefs to send fewer bytes over the bus):
- read_size - size of minimal read
- prog_size - size of minimal program

And the numbers you want big (allows littlefs to cache more device side):
- cache_size (= block_size, larger than block size has no benefit)
- lookahead_size (= disk_size / 8, larger than disk_size / 8 has no benefit, uses 1 bit per block)

It's also worth noting that littlefs has some performance issues with very large block sizes: https://github.com/littlefs-project/littlefs/issues/203

geky, 2021-08-03:
Rereading your issue, is it possible to query the erase size of different flash chips over something like SFDP? That would allow you to leverage what the flash chip provides:
https://www.macronix.com/Lists/ApplicationNote/Attachments/1870/AN114v1-SFDP%20Introduction.pdf

SadE54, 2021-08-03:
Hi , thx for the answer !
You're right the problem is the erase block size that is 64Ko basically for 8MB flash. Some chips sourced can erase 4kB subsector erase , while others  can erase sub sectors of 32kB .. And a last one that does not support the sub erase command :-/
I will look at SFDP , but probably serial jedec ID could be enough in my case. The last problem is that i don't have access to the full source of the HAL lib , only basic commands , including jedec id.
Thanks 

For lookahead_size do you mean disk_size in bytes or number of blocks ?

geky, 2021-08-03:
That's true, a lookup table for the IDs would avoid a lot of the complexity. Unfortunately parsing the SFDP tables gets very complicated. Here is roughly where it's implemented in Mbed OS:
https://github.com/ARMmbed/mbed-os/blob/master/storage/blockdevice/source/SFDP.cpp#L333

> For lookahead_size do you mean disk_size in bytes or number of blocks ?

Ah good catch, lookahead_size = number of blocks / 8

It is storing 1 bit per block, but is in units of bytes to be consistent with the other size/buffer fields.

SadE54, 2021-08-03:
I contacted the hal developper and he will add SFDP reading and sub sector erase. So it will be up to me to adapt according :)
Thanks for giving the link for SFDP , it will probably help me !

geky, 2021-08-03:
Ah nice!

SadE54, 2022-04-08:
Back to work with littlefs !
Sadly , for now i'm still stuck with 64KB erase size 😖
I'm doint some test on my PC using VS and I'm worried about the cache needed in my case.
I need cache size  == block size (so 64KB)
But it's 64KB for read cache + 64KB + few bytes for look ahead 😆
If I never need to write an read at the same time (in the same mount/amount session) , is it possible to avoid at least one buffer ?