littlefs-project/littlefs #587: A couple of questions...

mjbutsch, 2021-08-02:
lfs_config.block_cycles : Is this supposed to be the number of write/erase cycles for a block? Why does example have such a low value then? Any guidance for setting this value (use prog/erase limit from part data sheet; use 50% of limit / ???)?

lfs_config.sync : What is the purpose of this function? How should it be implemented? Is it just a "write"?

Thanks for the help!

geky, 2021-08-03:
Good questions, let me answer them backwards

> lfs_config.nnn_size : Any guidance for setting these values?

These mostly depend on the hardware and RAM you have available.

You generally want these to be as small as your storage can handle:
- read_size
- prog_size
- block_size

And you want these to be as large as you have RAM for
- cache_size
- lookahead_size

For now the best resource is probably to search through the existing issues on github. I have a TODO item to add a more thorough document, which I realize I am severely behind on, but it is what it is.

> lfs_config.sync : What is the purpose of this function? How should it be implemented? Is it just a "write"?

lfs_config.sync says "hey any writes I sent you really need to be written now!". In more complex block devices there can be layers of caching/reordering that avoid going all the way to disk until it's needed. On SD/eMMC this caching often even exists in special buffers on the storage itself. But if power gets cut unexpectedly, the data in these buffers can be lost.

So generally lfs_config.sync is used to flush buffers at every layer. LittleFS calls sync when it needs to ensure the device state is consistent, such as in lfs_file_sync, lfs_file_close, or most other metadata tampering functions.

I you're porting LittleFS to a raw block device, you can safely define lfs_config.sync to `return 0`. Though it would be worth reading through your devices datasheet to make sure there is no SYNC command required for consistency during power-loss.

> lfs_config.block_cycles : Is this supposed to be the number of write/erase cycles for a block? Why does example have such a low value then? Any guidance for setting this value (use prog/erase limit from part data sheet; use 50% of limit / ???)?

Ah no! 50% is a terrible idea. This is probably on me for choosing such an ambiguous name/having insufficient documentation.

block_cycles defines the number of cycles until a metadata block is moved. The more a metadata block is moved, the more distributed the wear.

Unfortunately, due to LittleFS design, moving metadata blocks is fairly expensive, so larger numbers means fewer dependent blocks must be written (and interestingly lower wear).

So 50% is a bad idea because littlefs would only move the metadata block after fully wearing the block %50 of its total lifetime. It would be better to choose a number around 10, 100 or 1000 cycles to minimize the wear to the blocks while also minimizing the number of metadata relocations.

Unfortunately I don't have much more info than that. The question of what is the best block_cycles is sort of unanswered. I know it's >2 and <n*erase_limit, but I haven't been able to test for what the optimal value would be.

mjbutsch, 2021-08-03:
Thanks for the reply.

I am using a SPI NOR device (MX25L3233F).

When you say:

> You generally want these to be as small as your storage can handle:
> 
> read_size
> prog_size
> block_size
> 

do you mean as small as the device can handle? The only "fixed" value seems to be the .block_size of 4096. I believe I can read/program a byte, but that seems bad. Just as using the block size of 4096 for read/prog seems wasteful of the RAM I have available. The example uses 16 for .read_size and .prog_size. Should they match the block size? or pick something that uses an amount of RAM I can live with (I presume powers of 2 are best)?

> And you want these to be as large as you have RAM for
> 
> cache_size
> lookahead_size

Again, using a block_size seems wasteful. Should these equal what is used for the read/prog size or be larger/smaller?

Thanks again!

geky, 2021-08-03:
I think some of the confusion is that only some of these fields have related RAM buffers. `read_size`/`prog_size`/`block_size` aren't used to allocate memory, they are only used to determine what the device is capable of. `cache_size`/`lookahead_size` on the other hand do use RAM buffers, so there is a trade-off between performance and memory, though they quickly have diminishing returns.

> I believe I can read/program a byte, but that seems bad.

Setting `read_size`/`prog_size` to a byte is correct. This is usually the case on NOR flash, it gets weirder on larger storage (NAND, eMMC, etc)

This tells LittleFS that it must read/write in multiples of that size. The key word there being "multiples". So if `read_size` = 1 byte, LittleFS may write 61 bytes depending on how much data is available. If `read_size` = 64 bytes, LittleFS will have to pad the data up to 64 bytes so that the driver can handle it.

If you have a loop like this in your driver you can batch together multiple bytes into single program commands to reduce bus overhead to the device:
https://github.com/ARMmbed/mbed-os/blob/master/storage/blockdevice/COMPONENT_SPIF/source/SPIFBlockDevice.cpp#L268

Usually these reads/writes are limited to the `cache_size`, but this isn't always the case, sometimes LittleFS can bypass the cache.

> Just as using the block size of 4096 for read/prog seems wasteful of the RAM I have available.

LittleFS doesn't need to allocate a buffer for the block_size. This number just tells LittleFS what unit to use for the blocks on the storage. This number can get quite large (~64KiB on NAND), so by necessity all flash filesystems must operate on smaller units inside the block.

> I presume powers of 2 are best

LittleFS doesn't actual need powers of 2, but it needs read/prog sizes to be factors of the block size, which gets complicated if you don't use powers of 2. But sometimes it can be useful, if you wanted to sneak ECC or hashes between pages on a block.

> Again, using a block_size seems wasteful. Should these equal what is used for the read/prog size or be larger/smaller?

This is a trade-off of performance vs memory, so it depends on your application. Last I checked Mbed OS defaults to 64 bytes, but this could be increased if your application is disk heavy:
https://github.com/ARMmbed/mbed-os/blob/master/storage/filesystem/littlefsv2/mbed_lib.json#L16

There are diminishing returns though, at some point throwing more RAM at the problem won't speed things up.

There are maximums, where more RAM just goes unused (actually should LittleFS assert on these?)
- cache_size <= block_size, only a single block can be cached at the time of writing
- lookahead_size <= block_count / 8, lookahead_size stores a bit per block

But the actual values you should use depend on your application + device + RAM. So the only real answer would be to test and see what you can get away with on your system.