littlefs-project/littlefs #590: nand flash ,when bad block at 0 or 1,littlefs don't work

zbqxyz, 2021-08-13:


zbqxyz, 2021-08-13:
// can't relocate superblock, filesystem is now frozen
if (lfs_pair_cmp(dir->pair, (const lfs_block_t[2]){0, 1}) == 0) {
LFS_WARN("Superblock 0x%"PRIx32" has become unwritable",
dir->pair[1]);
return LFS_ERR_NOSPC;
}

TomasPech, 2021-08-13:
Hi zbqxyz,
I have a same or similar problem and  yes, if problem is in 0,1 block then fails with LFS_ERR_NOSPC that is clear, if it is a bug or not I can't say.
Yesterday I started work on "Flash driver"  but I'm hanging there, on same problem as you, but I found that block isn't really damage, I will try to explain: raw writen data are matching with read data. But problem comes from CRC computation.
Here is my setup:

```
Flash_Erase_WHOLE_Chip();

	if(lfs_mount(&lfs, &cfg) != LFS_ERR_OK)
	{
		if(lfs_format(&lfs, &cfg) != LFS_ERR_OK)
		{
			__asm("nop"); // serious problem there ...
		}
	}

```

when chip is erased I'm not able to mount (of course) and flash need to be formated, but then fails on: LFS_ERR_NOSPC (traced). When I read raw data from Flash mem after failed lfs_format then I saw that address 0x00 contains value 0, rest of them was OK. Then I expected problem in driver, one dummy byte more is transfered during read process. As a solution I read one byte more (shifted first byte out), then I was able to sucessfully format Flash chip. After suceesfull format process was expected perfect function of "fs_mount"  but it failed again (CRC computation ). 
When flash was formated coretly and Flash_Erase_WHOLE_Chip() was comented out. then I removed "read extra dummy byte" from flash and then I was able to mount the flash ... 

I just **started** to find out this problem then I can't say something exact but tomorrow I will check if "lfs_format" giving right data to driver becasue all bytes in buffer from (lfs_format seems to be shifted to right side  byr one byte.

Please checkif is it also your problem....

zbqxyz, 2021-08-16:
> Hi zbqxyz,
> I have a same or similar problem and yes, if problem is in 0,1 block then fails with LFS_ERR_NOSPC that is clear, if it is a bug or not I can't say.
> Yesterday I started work on "Flash [...]

It should be a driver problem, you try to detach the littlefs, test whether the driver block read and write correct

TomasPech, 2021-08-17:
Hi,

> It should be a driver problem, you try to detach the littlefs, test whether the driver block read and write correct

Yes, that was driver, I didn't believe it was broken littleFS. The problem was crazy DMA - the DMA transfer request was hanging and then read the first "dummy" byte immediatelly after the stream was enabled. But that is happen only if another DMA transfer has beed made before... Also it never happened when I debugged by stepping, then all blocks were Written&Read correctly when I stepped the memory driver.  - Thank you STM32F4 with SPI and DMA2 for nice spend 4 hours 😄  - 

Anyway thank you ... 

Maybe we should delete my previous post becouse it has nothing to do with the issue opened by you.

HenryGiraldo, 2021-08-19:
> nand flash ,when bad block at 0 or 1,littlefs don't work

I also encountered this problem and it seems to be done that way by design. If you look at the badblocks test (test_badblocks.toml) right in the first test, bad blocks are evaluated one by one and the first 2 are skipped.

This makes me wonder... what would happen if during a normal use the first or the second block would be corrupted?

zbqxyz, 2021-08-20:
> 
> 
> > nand flash ,when bad block at 0 or 1,littlefs don't work
> 
> I also encountered this problem and it seems to be done that way by design. If you look at the badblocks test (test_badblocks.toml) right in the first test, bad blocks are evaluated one by one and the first 2 are skipped.
> 
> `for (lfs_block_t badblock = 2; badblock < LFS_BLOCK_COUNT; badblock++) {`
> 
> This makes me wonder... what would happen if during a normal use the first or the second block would be corrupted?

i test when bad block at 0/1,relocate block return below err 
// can't relocate superblock, filesystem is now frozen
if (lfs_pair_cmp(dir->pair, (const lfs_block_t[2]){0, 1}) == 0) {
LFS_WARN("Superblock 0x%"PRIx32" has become unwritable",
dir->pair[1]);
return LFS_ERR_NOSPC;
}