littlefs-project/littlefs #594: CRC Failure on Format?

tassociates, 2021-08-28:
Alcon,

I am working on trying to implement LittleFS on a Micron MT29F 8Gb chip  NAND.  

I can verify that my LLD for the chip can Read/Write/Erase blocks correctly -   I am trying to initialize the memory and format it, but keep getting the bad block on 0x0 error 

I can confirm that 0x0 is not bad, as the chip has ECC and the Bad Block flag in the spare area is 0xFF.

In trying to trace what is going on , I can see it writing a page of data that looks like the below and is 64 bytes in Length
0xff0xff0xff0xff0xff0xff0xff0xff0xff0xff0xff0xff0xff0xff0xff0xff0xff0xff0xff0xff0xff0xff0xff0xff0xff0xff0xff0xff0xff0xff0xff0xff0xff0xff0xff0xff0xff0xff0xff0xff0xff0xff0xff0xff

**0x700x1f0xfc0x80x990x380x140xc60xff**

0xff0xff0xff0xff0xff0xff0xff0xff0xff0xff0xff

I can confirm that reading this page back I get the same 64 byte values

However, when the code gets to the  **lfs_dir_commitcrc** function CRC check is failing on line 1436  of lfs.c

I dumped the output of the comparisons in the while loop - it checks 47 bytes and then exits 

DIR CRC WHILE BD READ ERR=0  COMMIT BLOCK=0x0 index=0, dat = 255 or 0xff CRC=16777215
DIR CRC WHILE BD READ ERR=0  COMMIT BLOCK=0x0 index=1, dat = 255 or 0xff CRC=65535
DIR CRC WHILE BD READ ERR=0  COMMIT BLOCK=0x0 index=2, dat = 255 or 0xff CRC=255
DIR CRC WHILE BD READ ERR=0  COMMIT BLOCK=0x0 index=3, dat = 255 or 0xff CRC=0
DIR CRC WHILE BD READ ERR=0  COMMIT BLOCK=0x0 index=4, dat = 255 or 0xff CRC=755167117
DIR CRC WHILE BD READ ERR=0  COMMIT BLOCK=0x0 index=5, dat = 255 or 0xff CRC=-1104801025
DIR CRC WHILE BD READ ERR=0  COMMIT BLOCK=0x0 index=6, dat = 255 or 0xff CRC=12461586
DIR CRC WHILE BD READ ERR=0  COMMIT BLOCK=0x0 index=7, dat = 255 or 0xff CRC=-558161693
DIR CRC WHILE BD READ ERR=0  COMMIT BLOCK=0x0 index=8, dat = 255 or 0xff CRC=350218095
DIR CRC WHILE BD READ ERR=0  COMMIT BLOCK=0x0 index=9, dat = 255 or 0xff CRC=-266646365
DIR CRC WHILE BD READ ERR=0  COMMIT BLOCK=0x0 index=10, dat = 255 or 0xff CRC=1647117971
DIR CRC WHILE BD READ ERR=0  COMMIT BLOCK=0x0 index=11, dat = 255 or 0xff CRC=1147535477
DIR CRC WHILE BD READ ERR=0  COMMIT BLOCK=0x0 index=12, dat = 255 or 0xff CRC=220793918
DIR CRC WHILE BD READ ERR=0  COMMIT BLOCK=0x0 index=13, dat = 255 or 0xff CRC=-328279254
DIR CRC WHILE BD READ ERR=0  COMMIT BLOCK=0x0 index=14, dat = 255 or 0xff CRC=-162183040
DIR CRC WHILE BD READ ERR=0  COMMIT BLOCK=0x0 index=15, dat = 255 or 0xff CRC=-1068746267
DIR CRC WHILE BD READ ERR=0  COMMIT BLOCK=0x0 index=16, dat = 255 or 0xff CRC=-39668413
DIR CRC WHILE BD READ ERR=0  COMMIT BLOCK=0x0 index=17, dat = 255 or 0xff CRC=-1037411054
DIR CRC WHILE BD READ ERR=0  COMMIT BLOCK=0x0 index=18, dat = 255 or 0xff CRC=-562449256
DIR CRC WHILE BD READ ERR=0  COMMIT BLOCK=0x0 index=19, dat = 255 or 0xff CRC=-754414257
DIR CRC WHILE BD READ ERR=0  COMMIT BLOCK=0x0 index=20, dat = 255 or 0xff CRC=-877479167
DIR CRC WHILE BD READ ERR=0  COMMIT BLOCK=0x0 index=21, dat = 255 or 0xff CRC=1523477920
DIR CRC WHILE BD READ ERR=0  COMMIT BLOCK=0x0 index=22, dat = 255 or 0xff CRC=-74546680
DIR CRC WHILE BD READ ERR=0  COMMIT BLOCK=0x0 index=23, dat = 255 or 0xff CRC=589490493
DIR CRC WHILE BD READ ERR=0  COMMIT BLOCK=0x0 index=24, dat = 255 or 0xff CRC=1967751541
DIR CRC WHILE BD READ ERR=0  COMMIT BLOCK=0x0 index=25, dat = 255 or 0xff CRC=219685823
DIR CRC WHILE BD READ ERR=0  COMMIT BLOCK=0x0 index=26, dat = 255 or 0xff CRC=1993431475
DIR CRC WHILE BD READ ERR=0  COMMIT BLOCK=0x0 index=27, dat = 255 or 0xff CRC=2132597986
DIR CRC WHILE BD READ ERR=0  COMMIT BLOCK=0x0 index=28, dat = 255 or 0xff CRC=1668902917
DIR CRC WHILE BD READ ERR=0  COMMIT BLOCK=0x0 index=29, dat = 255 or 0xff CRC=1561027186
DIR CRC WHILE BD READ ERR=0  COMMIT BLOCK=0x0 index=30, dat = 255 or 0xff CRC=-1823148801
DIR CRC WHILE BD READ ERR=0  COMMIT BLOCK=0x0 index=31, dat = 255 or 0xff CRC=9655540
DIR CRC WHILE BD READ ERR=0  COMMIT BLOCK=0x0 index=32, dat = 255 or 0xff CRC=-1747825956
DIR CRC WHILE BD READ ERR=0  COMMIT BLOCK=0x0 index=33, dat = 255 or 0xff CRC=-1561287880
DIR CRC WHILE BD READ ERR=0  COMMIT BLOCK=0x0 index=34, dat = 255 or 0xff CRC=94545840
DIR CRC WHILE BD READ ERR=0  COMMIT BLOCK=0x0 index=35, dat = 255 or 0xff CRC=-429457754
DIR CRC WHILE BD READ ERR=0  COMMIT BLOCK=0x0 index=36, dat = 255 or 0xff CRC=307335086
DIR CRC WHILE BD READ ERR=0  COMMIT BLOCK=0x0 index=37, dat = 255 or 0xff CRC=478032109
DIR CRC WHILE BD READ ERR=0  COMMIT BLOCK=0x0 index=38, dat = 255 or 0xff CRC=-207286408
DIR CRC WHILE BD READ ERR=0  COMMIT BLOCK=0x0 index=39, dat = 255 or 0xff CRC=1932505996
DIR CRC WHILE BD READ ERR=0  COMMIT BLOCK=0x0 index=40, dat = 255 or 0xff CRC=-914419915
DIR CRC WHILE BD READ ERR=0  COMMIT BLOCK=0x0 index=41, dat = 255 or 0xff CRC=2071483553
DIR CRC WHILE BD READ ERR=0  COMMIT BLOCK=0x0 index=42, dat = 255 or 0xff CRC=-1935145817
DIR CRC WHILE BD READ ERR=0  COMMIT BLOCK=0x0 index=43, dat = 255 or 0xff CRC=1698460098
DIR CRC WHILE BD READ ERR=0  COMMIT BLOCK=0x0 index=44, dat = 112 or 0x70 CRC=621473489
DIR CRC WHILE BD READ ERR=0  COMMIT BLOCK=0x0 index=45, dat = 31 or 0x1f CRC=2096751961
DIR CRC WHILE BD READ ERR=0  COMMIT BLOCK=0x0 index=46, dat = 252 or 0xfc CRC=-1497321854
DIR CRC WHILE BD READ ERR=0  COMMIT BLOCK=0x0 index=47, dat = 8 or 0x8 CRC=231451280
DIR COMPACT COMMIT CRC ERR=-84

Does anyone know what it is doing and why I would be getting a failure?

Many up front thanks

space-individual, 2021-09-01:
Hey tassociates,
I was running into a similar issue with our flash controller.  One thing to note is that littlefs seems to try to write pages before they are erased when it tries to lookahead.  This normally doesn't cause issues, but on the specific flash controller I was using, I ran into quite a few issues because on when a page is erased, that is when ECC is initialized on that block.  
As a result, when littlefs tried to read blocks that had not been erased, the flash controller would unsuccessfully try to correct those sections over and over again.  The solution for us was to assume that we would only get a certain amount of ECC errors per read, and ignore the errors if we got every single sector as needing correction.  I created a github issue post with the link here:

https://github.com/littlefs-project/littlefs/issues/573

Let me know if this helps you.  If you can, it may also be helpful to try manually reading, and erasing blocks of flash.  Try specifically reading an unerased section of flash, and see if you run into any issues.

Vokonn, 2022-03-02:
I have the same issue.

I have turned off ECC.

After the initial format, I am getting the following error messages.

> littlefs/lfs.c:1823:debug: Bad block at 0x0
> littlefs/lfs.c:1829:warn: Superblock 0x0 has become unwritable
> littlefs/lfs.c:4993:trace: lfs_format -> -28

so, I traced back and record data read from and written to NAND flash

The error was initated by lfs_dir_commitcrc()

> littlefs/lfs.c:1449:trace: Err lfs_dir_commitcrc CRC!=0 crc: -187111356 
> littlefs/lfs.c:1796:trace: err6 -84

```
        // detected write error?
        if (crc != 0) {
            return LFS_ERR_CORRUPT;
        }
```

so I traced back furthermore

> littlefs/lfs.c:5019:trace: lfs_mount(1db2, ffffffffffffcf04 {.context=0, .read=3ae, .prog=3b2, .erase=3b6, .sync=3ba, .read_size=64, .prog_size=64, .block_size=131072, .block_count=1024, .block_cycles=512, .cache_size=64, .lookahead_size=32, .read_buffer=1e6e, .prog_buffer=1eae, .lookahead_buffer=2f6c, .name_max=0, .file_max=1024, .attr_max=0}). 
> 
> fsRead PA: 0  _CA: 0 Read _Len= 64 
> 01 00 00 00 ff ff ff ff 6c 69 74 74 6c 65 66 73 2f f0 ff ff 02 00 00 00 02 00 00 00 00 04 00 00 ff 00 00 00 00 04 00 00 fe 03 00 00 fc 1f 00 00 ab ef ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
> 
> fsRead PA: 64  _CA: 0 Read _Len= 64 
> ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff
> 
> 
> littlefs/lfs.c:898:trace: lfs_dir_fetchmatch CRC0..idx:0 dat:01 CRCB=-1 CRCA=1526341860 .idx:1 dat:00 CRCB=1526341860 CRCA=-1489118143 .idx:2 dat:00 CRCB=-1489118143 CRCA=24923354 .idx:3 dat:00 CRCB=24923354 CRCA=1711753094 
> 
> fsRead PA: 0  _CA: 0 Read _Len= 64 
> 01 00 00 00 ff ff ff ff 6c 69 74 74 6c 65 66 73 2f f0 ff ff 02 00 00 00 02 00 00 00 00 04 00 00 ff 00 00 00 00 04 00 00 fe 03 00 00 fc 1f 00 00 ab ef ff ff ff ff ff ff ff ff ff ff ff ff ff ff
> 
> littlefs/lfs.c:918:trace: lfs_dir_fetchmatch CRC1..idx:0 dat:ff CRCB=1711753094 CRCA=700436191 .idx:1 dat:ff CRCB=700436191 CRCA=994549510 .idx:2 dat:ff CRCB=994549510 CRCA=-1000731353 .idx:3 dat:ff CRCB=-1000731353 CRCA=-1999896341 .
> littlefs/lfs.c:918:trace: lfs_dir_fetchmatch CRC1..idx:0 dat:6c CRCB=-1999896341 CRCA=1934940803 .idx:1 dat:69 CRCB=1934940803 CRCA=1085038524 .idx:2 dat:74 CRCB=1085038524 CRCA=-1778391331 .idx:3 dat:74 CRCB=-1778391331 CRCA=-1348475734 .
> 
> littlefs/lfs.c:992:trace: lfs_dir_fetchmatch CRC3..idx:0 dat:6c CRCB=-1348475734 CRCA=1923676257 littlefs/lfs.c:992:trace: lfs_dir_fetchmatch CRC3..idx:0 dat:65 CRCB=1923676257 CRCA=119500001 littlefs/lfs.c:992:trace: lfs_dir_fetchmatch CRC3..idx:0 dat:66 CRCB=119500001 CRCA=1943734767 littlefs/lfs.c:992:trace: lfs_dir_fetchmatch CRC3..idx:0 dat:73 CRCB=1943734767 CRCA=-104201114 littlefs/lfs.c:992:trace: lfs_dir_fetchmatch CRC3..idx:0 dat:2f CRCB=-104201114 CRCA=267989808 littlefs/lfs.c:992:trace: lfs_dir_fetchmatch CRC3..idx:0 dat:f0 CRCB=267989808 CRCA=-1687471229 littlefs/lfs.c:992:trace: lfs_dir_fetchmatch CRC3..idx:0 dat:ff CRCB=-1687471229 CRCA=1495815724 littlefs/lfs.c:992:trace: lfs_dir_fetchmatch CRC3..idx:0 dat:ff CRCB=1495815724 CRCA=528722744 littlefs/lfs.c:992:trace: lfs_dir_fetchmatch CRC3..idx:0 dat:02 CRCB=528722744 CRCA=-971810279 littlefs/lfs.c:992:trace: lfs_dir_fetchmatch CRC3..idx:0 dat:00 CRCB=-971810279 CRCA=1689107354 
> 
> littlefs/lfs.c:992:trace: lfs_dir_fetchmatch CRC3..idx:0 dat:00 CRCB=1689107354 CRCA=280942561 littlefs/lfs.c:992:trace: lfs_dir_fetchmatch CRC3..idx:0 dat:00 CRCB=280942561 CRCA=-685937607 littlefs/lfs.c:992:trace: lfs_dir_fetchmatch CRC3..idx:0 dat:02 CRCB=-685937607 CRCA=-1310919608 littlefs/lfs.c:992:trace: lfs_dir_fetchmatch CRC3..idx:0 dat:00 CRCB=-1310919608 CRCA=2025198934 littlefs/lfs.c:992:trace: lfs_dir_fetchmatch CRC3..idx:0 dat:00 CRCB=2025198934 CRCA=-2106572076 littlefs/lfs.c:992:trace: lfs_dir_fetchmatch CRC3..idx:0 dat:00 CRCB=-2106572076 CRCA=-2126748017 littlefs/lfs.c:992:trace: lfs_dir_fetchmatch CRC3..idx:0 dat:00 CRCB=-2126748017 CRCA=2105975511 littlefs/lfs.c:992:trace: lfs_dir_fetchmatch CRC3..idx:0 dat:04 CRCB=2105975511 CRCA=531039692 littlefs/lfs.c:992:trace: lfs_dir_fetchmatch CRC3..idx:0 dat:00 CRCB=531039692 CRCA=-1832048226 littlefs/lfs.c:992:trace: lfs_dir_fetchmatch CRC3..idx:0 dat:00 CRCB=-1832048226 CRCA=388330346 
> 
> littlefs/lfs.c:992:trace: lfs_dir_fetchmatch CRC3..idx:0 dat:ff CRCB=388330346 CRCA=-2139995464 littlefs/lfs.c:992:trace: lfs_dir_fetchmatch CRC3..idx:0 dat:00 CRCB=-2139995464 CRCA=-986035716 littlefs/lfs.c:992:trace: lfs_dir_fetchmatch CRC3..idx:0 dat:00 CRCB=-986035716 CRCA=-1261534082 littlefs/lfs.c:992:trace: lfs_dir_fetchmatch CRC3..idx:0 dat:00 CRCB=-1261534082 CRCA=-1224109377 littlefs/lfs.c:992:trace: lfs_dir_fetchmatch CRC3..idx:0 dat:00 CRCB=-1224109377 CRCA=1533650831 littlefs/lfs.c:992:trace: lfs_dir_fetchmatch CRC3..idx:0 dat:04 CRCB=1533650831 CRCA=2050044687 littlefs/lfs.c:992:trace: lfs_dir_fetchmatch CRC3..idx:0 dat:00 CRCB=2050044687 CRCA=-1866126174 littlefs/lfs.c:992:trace: lfs_dir_fetchmatch CRC3..idx:0 dat:00 CRCB=-1866126174 CRCA=944244712 littlefs/lfs.c:992:trace: lfs_dir_fetchmatch CRC3..idx:0 dat:fe CRCB=944244712 CRCA=-185795242 littlefs/lfs.c:992:trace: lfs_dir_fetchmatch CRC3..idx:0 dat:03 CRCB=-185795242 CRCA=469059974 
> 
[... cut ...]
> littlefs/lfs.c:1449:trace: Err lfs_dir_commitcrc CRC!=0 crc: -187111356 
> 

The read/write/erase function seems to be working.

I am getting the block with 'littlefs' strings.　　However, I am not sure if the rest of the block has the right content or format.

The block data read back from NAND flash is exactly what was written.  
I am new to file system.  I have been trying to figuring out what caused the problem, CRC!=0.

Can someone shad some light on this problem?

Any help would be appreciated.

marcosgabrielserrano, 2024-11-16:
@tassociates did you ever resolve this? 

I ran into similar error with Micron MT29F 2Gb.
I realize it's years later but mine occurs on 

/lfs.c:2077:warn: Superblock 0x0 has become unwritable
and ultimately on line 1765 of lfs.c: 
```
    // check non-padding commits against known crc
    if (crc != crc1) {
        return LFS_ERR_CORRUPT;
    }
```

MohamedAitB, 2025-01-21:
@marcosgabrielserrano hello do you resolve this ? i have a similar error

marcosgabrielserrano, 2025-01-25:
@MohamedAitB 

For me it turned out that my prog/read implementations were incorrect. I didn't understand how the plane bit worked on reads from the cache and progs. Double check those are working first.

Additionally follow the documentation to make sure there are no "bad blocks".