littlefs-project/littlefs #600: Corrupted dir pair at {0x1, 0x0}: Read error after successful writing

okosch, 2021-09-22:
Okay. Where to start.

I'm using littleFS with **Winbond W25N01GV Nand flash device**. In order to access the flash device read, write and erase functions have been created. 
All three functions have been tested without littleFS using both **Microchip dsPIC33CH512MP508** and **PIC32MZ2048EFH144**:
-	Reading pages
-	Writing pages
-	Erasing blocks
-	…
No issues so far.

After that littleFS is added with the following settings:

const struct lfs_config cfg = {
    // block device operations
    .read  = flash_read,
    .prog  = flash_write,
    .erase = flash_erase,
    .sync  = NULL,
    // block device configuration
    .read_size = 512,
    .prog_size = 512,
    .block_size = 2048UL * 64,
    .block_count = 1024 - 1,
    .cache_size = 512,
    .lookahead_size = 16,
    .block_cycles = 1000,
};

-	prog_size has been chosen to be 512 Bytes because the Flash device allows four partial programming accesses to each single page (2048 Bytes/page)
-	block_count is 1024 – 1 because I don’t want to use the first block of the Flash device. It contains the so called unique ID page and other information that can only be read. Due to this fact read, write and erase function contain an “offset” of one block.
-	lookahead_size is 16. **Can you confirm that this value is a valid one? I’m not sure whether I chose it correctly (found this description: lookahead_size <= block_count / 8)**

In order to verify the functionality of littleFS I’ve adapted the “boot_count” example: 
- After powerup the filesystem is mounted successfully. If mounting wasn’t possible formatting is executed. This happens only when the Flash device has been erased before.
- In an endless loop the following array is read, the content of each member incremented by one and after that the array is written. Only a single file is used.
uint8_t test_array[2100] = { 0 };
- If a read error occurs unmount followed by mount is executed.

**Using Microchip PIC32MZ2048EFH144 I haven’t observed any issues so far.** 

**But using Microchip dsPIC33CH512MP508 a sporadic error occurs:**

After many reading and writing cycles as mentioned above I get a read error. Please have a look at the following debug information:

lfs_file_opencfg(2DCE, 3DFC, "filename_1", 102, 3E4A {.buffer=3328, .attrs=0, .attr_count=0})
lfs_file_opencfg -> 0
lfs_file_write(2DCE, 3DFC, 35B8, 2100)
lfs_file_write -> 2100
lfs_file_close(2DCE, 3DFC)
lfs_file_close -> 0
// writing successful

lfs_file_opencfg(2DCE, 3DF8, "filename_1", 101, 3E46 {.buffer=3328, .attrs=0, .attr_count=0})
lfs_file_opencfg -> 0
lfs_file_read(2DCE, 3DF8, 35B8, 2100)
lfs_file_read -> 2100
lfs_file_close(2DCE, 3DF8)
lfs_file_close -> 0
// reading succesful

lfs_file_opencfg(2DCE, 3DFC, "filename_1", 102, 3E4A {.buffer=3328, .attrs=0, .attr_count=0})
lfs_file_opencfg -> 0
lfs_file_write(2DCE, 3DFC, 35B8, 2100)
lfs_file_write -> 2100
lfs_file_close(2DCE, 3DFC)
lfs_file_close -> 0
// writing successful

lfs_file_opencfg(2DCE, 3DF8, "filename_1", 101, 3E46 {.buffer=3328, .attrs=0, .attr_count=0})
Corrupted dir pair at {0x1, 0x0}
lfs_file_opencfg -> -84
// reading error

lfs_unmount(2DCE)
lfs_unmount -> 0
// unmount successful 

lfs_mount(2DCE, 2E3C {.context=2DCE, .read=B85A, .prog=AA68, .erase=E98A, .sync=EA2C, .read_size=512, .prog_size=512, .block_size=131072, .block_count=1023, .block_cycles=1000, .cache_size=512, .lookahead_size=16, .read_buffer=2EF0, .prog_buffer=3100, .lookahead_buffer=3308, .name_max=0, .file_max=0, .attr_max=0})
Corrupted dir pair at {0x1, 0x0}
lfs_mount -> -84
// mount not successful

- The data of the used file cannot be accessed at all. 
- Formatting creates a new file system successfully (including data loss). 
- Read, write and erase functions also create debug entries if an error occurs. But entries from these functions haven’t been observed.
- According to the debug log the last writing was successful. After that the next attempt to read delivers -84 (LFS_ERR_CORRUPT). 

**So here is my biggest question: Is this error/behavior related to the fact that I’m using a 16 Bit controller?**

Many thanks in advance for your hints, explanations and solutions. 
Looking forward to hearing from you.

okosch, 2021-09-23:
Here’s a short update:
I’ve adjusted my code and added more debug log calls to it including debug information of every single read, write and erase access to my Flash device.
In the following description Block numbers are mentioned. These numbers refer to the LittelFS system.

This is what I observe:
-	Block 0 and 1 are used in order to store the so called metadata pairs
-	Actual data is stored to free blocks (Block 0 and 1 are not touched for actual data)
-	Block 0 and 1 are only erased during compaction

-	After several hours the following situation occurs:
o	Metadata is updated to Block 0
o	**Block 0 is erased and actual data is stored there**
o	Metadata is tried to be read from Block 0. This seems not possible. Therefore metadata is extracted from Block 1. This is successful. 
Nevertheless some of the history stored in Block 0 is lost. 
As a consequence my data is older. 
After this hiccup my program continues with periodically reading and writing data as mentioned in my first entry to this issue ( (see above: https://github.com/littlefs-project/littlefs/issues/600#issue-1003984570)
-	A couple of hours later the strongest hiccup occurs:
o	Metadata is updated to Block 1
o	**Block 0 is erased and actual data is stored there**
o	Metadata is still updated to Block 1
o	**Now Block 1 is erased and actual data is stored there**

**_Now both the latest metadata and the “backup” are lost
After that every access to the filesystem results in -84 (LFS_ERR_CORRUPT)_**

As a next step I’m trying to check the block allocator.

okosch, 2021-09-27:
Hi @geky 
I was checking block allocator code:

-	lfs_alloc_lookahead
-	lfs_alloc

Both functions contain shift left operations (maximum 31 shifts) using “1U”. For 16bit systems 1U is represented by a 16bit value. 
I added a cast and started my test again. 
Result: even after 5 days of continuous reading/writing to the file system the error mentioned above did not occur. 

**Main concerns/questions:**
-	**LittleFS 16 Bit support: YES/NO?**
-	**(I will continue my integration tests, but) are you aware of further variable declarations/usage/pitfalls similar to the ones included in block allocator?**
-	**Do you know other users who have successfully used LittleFS on 16bit machines?**

Vokonn, 2022-03-02:
Hi @okosch  
I am using similar hardware, Winbond W25N01GV and Microchip dsPIC33EP512MU814.
When I tried to format FS for the first time, I have been getting CRC error.  
#594 (trace detail were posted here)
Read/Prog/Erase functions are working properly.
Did you encounter similar issue?
I am really struggling.  
Any help would be appreciated.