littlefs-project/littlefs #603: problem with LFS_NO_MALLOC defined

vrz, 2021-10-05:
with LFS_NO_MALLOC defined problem happend when open read file.
i think that problem is when calling lfs_file_rawopencfg() with empty cfg.
string 2624
"    int err = lfs_file_rawopencfg(lfs, file, path, flags, &defaults);"

nnayo, 2021-11-09:
I have the same issue.
the call stack is:
lfs_file_open() -> lfs_file_rawopen() -> lfs_file_rawopencfg()

struct lfs_file_config defaults = {0} is set in lfs_file_rawopen() just before calling lfs_file_rawopencfg().
this is fine expect when LFS_NO_MALLOC is defined because lfs_malloc() is nevertheless called in lfs_file_rawopencfg() and returns NULL (see lfs_util.h).

file->cache.buffer seems to be needed as long as the file is open, seeing lfs_file_rawclose() calling lfs_free().

assigning a static buffer to file->cache.buffer seems to solve the problem as long as only one file is open at a time.

my concern is: can the cache be reused / shared for other files ?

mfarver, 2021-11-10:
I made it work using lfs_file_opencfg and passing in a lfs_file_config structure that pointed at a static file buffer:
```
static lfs_t lfs;
static uint8_t FileCache[ 16 ] __attribute__ ((aligned (32)));
const struct lfs_file_config Fileconfig = {
    .buffer = &FileCache,
};
.
.
.
lfs_file_opencfg(&lfs, &file, "boot_count", LFS_O_RDWR | LFS_O_CREAT,
        &Fileconfig);
```

nnayo, 2021-11-10:
Thanks,

That's more or less what I had in mind.

But in the LFS_NO_MALLOC case, lfs_file_open() should not be available then.

Also, in your case, the buffer is only 16 bytes long, does it match with your cache size?