littlefs-project/littlefs #608: v2.2.1 block tail points to itself ?

XinStellaris, 2021-11-08:
I have a lfs_mount problem. the lfs_mount will return LFS_ERR_CORRUPT in the end. 
After some debugging, I found that in meta-pair {2,3}, block 2 has a LFS_TYPE_TAIL tag, and it contains tail 2 and 3, whch are the 
 current meta-pair themselves.

XinStellaris, 2021-11-25:
Okay, I found the bug myself.
    Assume there are 6 blocks in littlefs, block 0,1,2,3  are in use. 0 has a tail pair of {2, 3}. Now we try to write more into 0.
    When writing to block 0, we will split(FIRST SPLIT), thus allocate block 4 and 5. Up to now , everything is as expected.
    Then we will try to commit in block 4, during which split(SECOND SPLIT) is triggered again(In our case, some files are large, some are small, one split may not be enough).  Still as expected now.
    BUG happens when we try to alloc a new block pair for the second split:
    As lookahead buffer reaches the end , a new lookahead buffer will be generated from flash content, and block 4, 5 are unclaimed blocks because they are not programed yet. HOWEVER, block 4,5 should be occupied in the first split!!!!!

XinStellaris, 2021-11-25:
I am at lost when I try to fix this. Is there any good idea on how to fix this?

XinStellaris, 2021-11-29:
It seems, that commit ce2c01f098f4d2b9479de5a796c3bb531f1fe14c results in this. In this commit, a lfs_alloc_ack is inserted in lfs_dir_split, which will cause split to reset lfs->free.ack to block count.
In summary, this problem exists after 2.1.3.

ysy4tc3, 2022-06-06:
I'm interesting in this issue. But I seems hard for me to reproduce it. Can I have a guidance to reproduce it?

XinStellaris, 2022-06-08:
> I'm interesting in this issue. But I seems hard for me to reproduce it. Can I have a guidance to reproduce it?

I have fixed this. see https://github.com/littlefs-project/littlefs/pull/620
Wish this helps.

ysy4tc3, 2022-06-08:
我已收到您的邮件，我一定会尽快处理。叶思远

closed 2022-06-10 by XinStellaris

ysy4tc3, 2022-06-13:
thx a lot