littlefs-project/littlefs #622: A question about lookahead_size

Johnxjj, 2021-12-08:
hi，@geky, What will happen if the number allocated by lookahead_size is less than block_count/8? Will it cause the file system to crash or the file to be written incorrectly?

Johnxjj, 2021-12-08:
I read the code and found that reading and writing lfs->free.buffer will limit the scope.
write：
     if (off <lfs->free.size) {
         lfs->free.buffer[off / 32] |= 1U << (off% 32);
     }

read：
         while (lfs->free.i != lfs->free.size) {
         read.......
     }

I have a problem now that when reading the CTZ list of a file, I find that the head pointer of a  block points to the super block, causing the read address to crash. I am not sure if it is because my lookahead_size allocation is less than block_count/8.

Johnxjj, 2021-12-13:
#define LFS_BLOCK_CYCLES         100
#define LFS_CACHE_SIZE           512
#define LFS_LOOKAHEAD_SIZE       32
#define LFS_READ_SIZE            128           
#define LFS_PROG_SIZE            128            
#define LFS_FLASH_SECTOR_SIZE    4096 
block_count = 413
lookahead_size = 32
[data.txt](https://github.com/littlefs-project/littlefs/files/7700557/data.txt)

Johnxjj, 2021-12-13:
#define LFS_VERSION 0x00020001

closed 2023-11-09 by Johnxjj

geky, 2023-11-14:
Hi @Johnxjj, sorry about not being able to keep up with all of these issues.

In general I've not been able to keep up with the amount of issues coming into littlefs.

For this issue specifically you may have noticed an assert on `lookahead_size % 8`, this is because the current code expects the lookahead size to be 8-byte aligned. This isn't strictly required, but the current code assume lookahead size is 8-byte aligned, and it's not been a priority to loosen this restriction.