littlefs-project/littlefs #625: On File Open how can you get starting block of File

mjs513, 2021-12-12:
We have been using LFS and have it successfully working on several different NOR chips and NAND chips.  Now we are trying to implement MTP on a Teensy micro but ran into a problem with one particular NOR Flash (W25Q512) and seems that it just not fast enough, i.e, format+write cycles.  One of the things we were thinking about was to format the blocks for the file prior to actually coping the file from the PC to the flash but to do this we need to know the starting block of the file. 

Guess the question is what do I have to do to get the starting block of the file from LFS?

geky, 2022-04-06:
Hi @mjs513, thanks for opening an issue and sorry about the late response.

The idea to format before download is quite a creative one, unfortunately I'm not sure it's one littlefs is well engineered for. 

> what do I have to do to get the starting block of the file from LFS?

This is unfortunately not really exposed and has a few challenges:

1. Files are copy-on-write, aka they move when written. This means the start of the file isn't decided until writing starts. If you somehow disabled this unfortunately the filesystem would no longer be power-resilient.

2. Files are broken into multiple blocks across disk, so you would need all of the blocks composing the file.

If you only need to update one file, you could consider using a partition table such as an [MBR](https://en.wikipedia.org/wiki/Master_boot_record), effectively storing the file next to littlefs. This would sacrifice wear-leveling, but if the file is written relatively infrequently (such as firmware updates) this may be ok.

geky, 2022-04-06:
Thinking about it more, it could be possible to add an additional API to allow files to be "pre-erased" in preparation for new data. But tracking this additional state would be rather complicated and involved, and likely increase the code-size for normal file writes.

It would be interesting to explore but probably low in priority on littlefs's roadmap.