littlefs-project/littlefs #627: Smallest reasonable FS

perigoso, 2021-12-16:
I have a MCU with an internal 1KB of EEPROM, 4byte page size, for reasons that dont really matter here I want/need to use lfs, what are the limitations here? I read something about operating in 2 block mode here, still i have 1/4 the memory than in this issue #506

e107steved, 2021-12-17:
You can set sector sizes down to 128 bytes minimum. So (IIRC - it's a long time since I set things up) you can set an erase block size of 128 bytes. I think it should work; well worth a try.  (I'm using it with 128K of FRAM, and there's minimal overhead, unlike FAT. Can't find my notes at present, though)

jrast, 2022-01-07:
You may also find some more detailed information about the minimal block size in this (very old) issue: https://github.com/littlefs-project/littlefs/issues/264

closed 2022-01-07 by perigoso