littlefs-project/littlefs #629: littlefs speed test

xinglongzhu, 2022-01-05:
hi

Do you have littlefs speed test document?

geky, 2022-02-19:
Hi @xinglongzhu, no, this is an area that needs improvement.

Speed has not been a priority, though that will be changing

anujdeshpande, 2024-04-15:
@geky do you have something in mind for how a speed test (or documentation) would look like? 
I have been investigating slow operations and would love to contribute in any way I can. I checked out your [benchmarks](https://github.com/geky/littlefs-benchmarks) repo - maybe that's a good start?

geky, 2024-04-16:
Hi @anujdeshpande, no not really to be honest.

There is a [large branch of work](https://github.com/littlefs-project/littlefs/tree/rbyd) ongoing that is focused on improving performance, but this is mostly at the algorithm level, so more math than benchmarking.

I think eventually benchmarks will have a role for optimizing configurations for specific storage types (NOR, NAND, SD, NVRAM, etc, these all behave slightly differently), but right now that's not a priority.

Oh and also for evaluating more advanced caching features, but that's also way out of scope right now.

Right now the [benchmark](https://github.com/geky/littlefs-benchmarks) repo has just been useful for verifying that algorithms aren't broken. Fortunately user feedback/measurements have been very useful and I think have identified most of the existing bottlenecks.