littlefs-project/littlefs #642: LittleFs Create Failed! arduino

Michal-Dudcenko, 2022-02-12:
I'm creating a WebLamp but when i upload certs.ar ,,175KB"" it say 
LittleFS Create Failed!
[LittleFS] data    : D:\Users\asusd\OneDrive\Рабочий стол\WebLamp\data
[LittleFS] size    : 64
[LittleFS] page    : 256
[LittleFS] block   : 4096
/certs.ar
lfs warn:475: No more free space 20
lfs_write error(-28): File system is full.

error adding file!

LittleFS Create Failed!

But in Tools > Flash size I set 1MB (FS:64KB OTA:~470KB
[image]
)

geky, 2022-02-19:
Hi @Michal-Dudcenko, it may be better to open this issue on the Arduino repo:
https://github.com/esp8266/Arduino/issues

Without more low-level details it's not really possible for us to debug this.

Michal-Dudcenko, 2022-02-19:
I generate certs.ar and put it to project folder > data

Michal-Dudcenko, 2022-02-19:
There is the file.
https://drive.google.com/file/d/15lJUT7c8eb2ghs6npbiOMXzhq8UNWJU-/view?usp=sharing

I paste google drive link beacouse github doesn't support .ar files

closed 2022-07-03 by Michal-Dudcenko