littlefs-project/littlefs #649: What does the x represent in CTZ skip-lists block/pointer relation equation

tony-josi, 2022-02-23:
Im having trouble figuring out what does the x represent in the following line from the [design ](https://github.com/littlefs-project/littlefs/blob/master/DESIGN.md) file. 

> The rules CTZ skip-lists follow are that for every n‍th block where n is divisible by 2‍ˣ, that block contains a pointer to block n-2‍ˣ. This means that each block contains anywhere from 1 to log₂n pointers that skip to different preceding elements of the skip-list.

Thanks

geky, 2022-02-24:
That part is a bit dense and handwavey isn't it.

x is a new variable that takes the value of any integer that satisfies n % (2^x) = 0 (I'm using a programmer's [modulus](https://en.wikipedia.org/wiki/Modulo_operation) here). Which is one way to write any integer x where n is divisible by 2^x.

```
.--------.  .--------.  .--------.  .--------.  .--------.  .--------.
| data 0 |<-| data 1 |<-| data 2 |<-| data 3 |<-| data 4 |<-| data 5 |
|        |<-|        |--|        |<-|        |--|        |  |        |
|        |<-|        |--|        |--|        |--|        |  |        |
'--------'  '--------'  '--------'  '--------'  '--------'  '--------'
```

- data 0 where n = 0, this is a special case, ignore this one
  no pointers (we need to stop somewhere!)
- data 1 where n = 1 -> x = 0 because 1 % 2^0 = 1 % 1 = 0
  so one pointer
- data 2 where n = 2 -> x = 0 or 1, because 2 % 2^0 = 2 % 1 = 0 and 2 % 2^1 = 2 % 2 = 0
  so two pointers
- data 3 where n = 3 -> x = 0, because 3 % 2^0 = 3 % 1 = 0
  so one pointer
- data 4 where n = 4 -> x = 0 or 1 or 2, because 4 % 2^0 = 4 % 1 = 0 and 4 % 2^1 = 4 % 2 = 0 and 4 % 2^2 = 4 % 4 = 0
  so three pointers
- data 5 where n = 5 -> x = 0, because 5 % 2^0 = 5 % 1 = 0
  so one pointer

What happens in the code is littlefs tries each value until this test fails. This determines how many pointers to add to the blocks. We made up this function because it has the nice properties outlined in that section such as needing only amortized 2 pointers per block while still allowing log-runtime traversal to anywhere.

tony-josi, 2022-02-24:
That's a cool idea. Thanks for the great explanation and amazing work on LittleFS. 
Are you considering adding this to the design markdown file as it might help understand the concept better?

geky, 2022-03-24:
Probably not anytime soon, just due to timing constraints and personal priorities.

Feel free to create a PR if you think it would be useful. You may actually be in a better position to word best for another reader to understand.