littlefs-project/littlefs #654: LittleFS configuration and memory allocation

deslim01, 2022-03-08:
Hi experts,

I have a question regarding the LittleFS memory configuration. Below is my setting of the memory. I also put in my comments in each line of the configuration for how much the memory is needed.

    .read_size = 16,     // Allow reading INT. My understanding is that NOR flash can perform byte-wise reading too. If
                                  // I want to perform byte reading, this number should be 8, am I correct?

    .prog_size = 256,  // This one should be fine. Can write 256 bytes.
    .block_size = 4096, // In NOR flash term, this should be Sector Size right?? One sector in NOR is about 4K. I also see that this is being set to 1024, I assume this should not be a problem for NAND flash right?

	.block_count = 256, // There are 256 blocks in total. Hence, 4K * 256 = 1M. From my understanding, that is equivalent to Sub-Block of the NOR-flash.

	.cache_size = 256, // Cache is 256 bytes..

	.block_cycles = 200, // For durability.
	
	.lookahead_size = 16 // Mainly for Cache usage.

Can someone comment if I only allocate 1M for NOR flash because I wanted to  use the other memory location for something else, it is considered enough using the configuration above?

geky, 2022-03-24:
Hi @deslim01,

I'm a bit confused by this line

> Can someone comment if I only allocate 1M for NOR flash because I wanted to use the other memory location for something else, it is considered enough using the configuration above?

This looks like a correct configuration, except for a note below. If it is enough depends on what you use the filesystem for. You can't store more files than you have flash for, for example.

---

```
.read_size = 16, // Allow reading INT. My understanding is that NOR flash can perform byte-wise reading too. If
                 // I want to perform byte reading, this number should be 8, am I correct?
```

The units here are in bytes, so `read_size = 1` is probably what you want. Some storage doesn't support byte-wise reading, which is why you might make this larger.

```
.prog_size = 256,  // This one should be fine. Can write 256 bytes.
```

Most NOR-flash also supports byte-wise writing, so you could try setting `prog_size = 1` as well. LittleFS uses these numbers for alignment/padding, it will still try to write as many bytes at once to take advantage of caches that might be on the device.

```
.block_size = 4096, // In NOR flash term, this should be Sector Size right?? One sector in NOR is about 4K. I also see that this is being set to 1024, I assume this should not be a problem for NAND flash right?
```

Yes, this is sector size in NOR-flash. Frustratingly, different storage devices use wildely different, and sometimes conflicting, names for read/write/erase blocks. Though the `block_size` name doesn't really help.

The `block_size` here is the unit of "blocks" in littlefs. It needs to be > the erase size (usually 4KiB on NOR-flash), but can be larger. Larger `block_size` can make some operations more efficient, but wastes more space.

I'm not sure about the 1024 number, NAND-flash goes up into 32 KiB/64 KiB erase block sizes.

deslim01, 2022-03-24:
Thank @geky  for your reply.

The reason I am asking this is because I am not really sure that how much memory needed for LittleFS say having 10000 1-byte file storage before it started crashing.

For now, I am just experimenting using 1M byte block for testing and given the configuration I set above, I was actually wondering if LittleFS will complain due to some minimum space required for such configuration or not.

Furthermore, my understanding that LittleFS needs space to do wear leveling. I was wondering what is the minimum space required for that as well.

.read_size = 16 ==> means 16 bytes in one read, am I right on that?
.prog_size = 1 ==> do you mean program 1 byte for each cycle? I understand the padding part but if it is set to 1 byte, I am a bit confused how it managed to work if the data is in double or int? Can you please elaborate more? Are there any document explaining how these configuration should be done properly and is set at its optimum performance?
.block_size ==> Understand. Hence this is defined as the minimum block based on the type of flash used. For NOR flash. this is set to 4KiB.

That's all from me.

Thanks.

geky, 2022-03-24:
> .prog_size = 1 ==> do you mean program 1 byte for each cycle? I understand the padding part but if it is set to 1 byte, I am a bit confused how it managed to work if the data is in double or int? Can you please elaborate more?

Ah, so `read_size` and `prog_size` set the _minimum_ amount of bytes that can be read or written in a single operation. LittleFS will use this for alignment, but may read or write any size up to the `block_size`. Reads or progs may be a _multiple_ or `read_size` and `prog_size`.

There is no way to limit the number of byte read or programed in a single `read`/`prog` call. The assumption is that your driver can take larger amounts of data and read/write the using the smaller size in a loop.

For example:
https://github.com/ARMmbed/dataflash-driver/blob/master/DataFlashBlockDevice.cpp#L377

> .read_size = 16 ==> means 16 bytes in one read, am I right on that?

To be clear, this means 16 bytes or more, in one read. But the read will always be in multiples of 16 bytes and aligned to 16 bytes.

> Furthermore, my understanding that LittleFS needs space to do wear leveling. I was wondering what is the minimum space required for that as well.

This doesn't have a single answer, it depends on how long you want your storage to last. My rule of thumb is 2x the original requirements, because that gives you plenty of space to work with and storage is cheap.

> Are there any document explaining how these configuration should be done properly and is set at its optimum performance?

There is some minimal documentation in lfs.h, but it is very minimal unfortunately.