littlefs-project/littlefs #655: cppcheck error in functions lfs_rawremove() and lfs_rawrename()

fuc2wi, 2022-03-08:
Hello.
In lfs v2.4.1 release I get the cppcheck error "Address of local auto-variable assigned to a function parameter" in functions lfs_rawremove() and lfs_rawrename() (line 3237 and 3349)
I was not able to analyse it completely (means, I don't know for what this pointer of the local variable will be used), but maybe this can be a problem?

geky, 2022-03-24:
That is quite an interesting lint, saving a pointer to a local variable is a pretty dangerous operation. It's easy to do accidentally in C and can lead to stack corruption.

But we're actually doing this here intentionally.

In littlefs, due to wear-leveling and other things, any operation can affect multiple metadata blocks. In the code this can become tricky to manage, as each open file or partial operation has its own copy of the metadata block's state in the `mdir` struct it's operating on.

To solve this, we keep track of all active `mdir` structs in a big linked-list called the `mlist` (because naming things is hard). Any metadata updates done in the lower-layers goes through this linked-list and makes sure any relevant `mdir`s are updated. Keeping the full system in sync.

Most stack-allocated `mdir` structs don't need to go into the `mlist` because they don't live long enough for other operations to occur (keep in mind littlefs is single-threaded). But `lfs_rename` and `lfs_remove` are tricky operations that may need to update several metadata blocks, at which point seemingly unrelated `mdir` updates can affect each other. So in these cases we do need to insert the stack-allocated `mdir` into the linked-list, though only temporarily.

The important, and error-prone, thing is that we restore the `mlist` to its original state before the function returns. For every possible return path:
- https://github.com/littlefs-project/littlefs/blob/ead50807f1ca3fdf2da00b77a0ce02651ded2d13/lfs.c#L3244
- https://github.com/littlefs-project/littlefs/blob/ead50807f1ca3fdf2da00b77a0ce02651ded2d13/lfs.c#L3248
- https://github.com/littlefs-project/littlefs/blob/ead50807f1ca3fdf2da00b77a0ce02651ded2d13/lfs.c#L3366
- https://github.com/littlefs-project/littlefs/blob/ead50807f1ca3fdf2da00b77a0ce02651ded2d13/lfs.c#L3378
- https://github.com/littlefs-project/littlefs/blob/ead50807f1ca3fdf2da00b77a0ce02651ded2d13/lfs.c#L3383

This would be a nice thing for cppcheck to check for, but I imagine it's a very hard problem.

This solution potentially wanders into "hack" territory, but it's a very efficient way to manage the state of the system.

fuc2wi, 2022-03-24:
Thanks for the detailed explanation.
So I will mark this as false positive in my cppcheck settings to suppress the error message.

closed 2022-03-24 by fuc2wi