littlefs-project/littlefs #661: littlefs storage utilization

dorsadeh, 2022-04-03:
I'm using littlefs on a 128MB external NAND flash
the flash erase resolution is only in blocks sized 128KB 
My application writes files ranging from 2KB - 60KB
I noticed littlefs writes every new file to a new flash block. in case of a 2KB file it is only 1.5625% utilization of block space!
Is this issue solvable?

geky, 2022-04-06:
Hi @dorsadeh thanks for opening an issue.

So littlefs can store multiple files in a block but it has a caveat that these files, called "inlined files", need to fit in the device's RAM, controlled by the cache_size configuration. So if you set cache_size >= 2KB, you should no longer see one block per file.

This is a tradeoff necessary to allow multiple open files, since a sync of any file might need to rewrite that block, possibly invalidating other open-but-unsynced files.

rabbitsaviola, 2022-04-19:
Hi @geky , it seems inline file size is limited by 0x3fe byte besides cache size. Even if cache size is increased to 32768,  2KB file would still use a new block. Is my understanding right?

```c
static lfs_ssize_t lfs_file_flushedwrite(lfs_t *lfs, lfs_file_t *file,
        const void *buffer, lfs_size_t size) {
    const uint8_t *data = buffer;
    lfs_size_t nsize = size;

    if ((file->flags & LFS_F_INLINE) &&
            lfs_max(file->pos+nsize, file->ctz.size) >
            lfs_min(0x3fe, lfs_min(
                lfs->cfg->cache_size,
                (lfs->cfg->metadata_max ?
                    lfs->cfg->metadata_max : lfs->cfg->block_size) / 8))) {
        // inline file doesn't fit anymore
        int err = lfs_file_outline(lfs, file);
        if (err) {
            file->flags |= LFS_F_ERRED;
            return err;
        }
    }

```

geky, 2022-04-20:
Ah, @rabbitsaviola, you are right, I had forgotten about the field limit. On disk inline files are stored with a 10-bit length field, with 0x3ff representing a deleted file, so there is a hard limit at ~1 KiB.

There _are_ 2 extra reversed bits in the tag (it wasn't clear if these 2 bits would be more useful for ids (files per metadata block) or for the attribute length). This could raise the limit to 12-bits (~4KiB) at the cost of some complexity since these bits aren't contiguous in the tag.

rabbitsaviola, 2022-04-21:
Thanks for your suggestion, @geky. I do tried Dhara as FTL for fatfs before. But its performance is much worse than littlefs. Metadata cache might be needed to improve the performance. I'll try it with littlefs to double check it.

kyrreaa, 2024-02-08:
With inline enabled it seems to be no way to actually utilize the first inline tag written as it is written during creation of the file upon open, and no writes to the file has happened yet. It would be very useful to be able to write both the name and the first inline data bit in that first record as it would be able to contain the initial header of my files.

_Examining block 64 with revcount = 00000002:
superblock name: littlefs
  inline struct: Ver:00020001  bsize:00040000  bcount:00001000  name_max:000000FF  file_max:7FFFFFFF  attr_max:000003FE
Offset 00262188: CRC block with padding. Length=1022
Offset 00263214: CRC block with padding. Length=1022
Offset 00264240: CRC block with padding. Length=1022
Offset 00265278: CRC block with padding. Length=958
Examining block 65:
Offset 00266240: id=001 Creating file:
   regular file: id=001 name='test1.txt'
    inline data: id=001 length=000 (0)
Offset 00266261: CRC block with padding. Length=1022
Offset 00267287: CRC block with padding. Length=1022
Offset 00268313: CRC block with padding. Length=1022
Offset 00269351: CRC block with padding. Length=981
Examining block 66:
    inline data: id=001 length=040 (64)
Offset 00270404: CRC block with padding. Length=1022
Offset 00271430: CRC block with padding. Length=1022
Offset 00272456: CRC block with padding. Length=1022
Offset 00273494: CRC block with padding. Length=934_

kyrreaa, 2024-02-09:
> I'm using littlefs on a 128MB external NAND flash the flash erase resolution is only in blocks sized 128KB My application writes files ranging from 2KB - 60KB I noticed littlefs writes every new [...]

There is also the config prog_size that is normally used for the sector size your flash can write, but as long as block_size is correctly defined erases will happen aligned correctly. I limited my metadata_max to reduce the time it takes for compaction (and to make it happen often) and ran a test on my nand flash with reduced prog_size after checking with my device datasheet.
My flash (mt29f8g) has 4096 byte "sectors" and 64*4096 byte eraseable block size, but it also mentions a maximum of 4 partial writes per "sector" allowed when using it's built in parity system. (Writing only the 4096 byte data area, and leaving it to automatically fill in the 256 byte metadata area.) I thus set up a test with cache_size and read_size equal to the sector size of 4096 and setting prog_size to read_size/4 for 1024 byte chunks. I first checked that my nand flash implementation would allow this and added code to check against any writes crossing the actual sector boundaries; splitting these into two or more operations. (Using larger cache would allow littlefs to write multiple sectors at once and I didn't want it to end up trying to write 2 sectors at once, offset by 1/4 sector...)
Running my file-creation and deletion test I could see a quite significant change in behaviour allowing littlefs to pack the initial metadata 4x tighter.
[image]
[image]

A quick examine of the filesystem after creating 5 files show how some of them are now packed better into each sector:
_Examining sector 65:
Offset 00041000: id=002 Creating file. Tag length=4
Offset 00041004: Regular file: id=002 name='test_001.txt'  Tag length=16
Offset 00041014: Inline data. id=002 length=000 (0) Tag length=4
Offset 00041018: CRC block with padding. Tag length=12
Offset 00041024: CRC block with padding. Tag length=988
Offset 00041400: Inline data. id=002 length=010 (16) Tag length=20
Offset 00041414: CRC block with padding. Tag length=12
Offset 00041420: CRC block with padding. Tag length=992
Offset 00041800: CTZ struct: id=002 head=00000A8C (2700) size=00000FA0 (4000) Tag length=12
Offset 0004180C: CRC block with padding. Tag length=12
Offset 00041818: CRC block with padding. Tag length=1000
Offset 00041C00: id=003 Creating file. Tag length=4
Offset 00041C04: Regular file: id=003 name='test_002.txt'  Tag length=16
Offset 00041C14: Inline data. id=003 length=000 (0) Tag length=4
Offset 00041C18: CRC block with padding. Tag length=12
Offset 00041C24: CRC block with padding. Tag length=988
Examining sector 66:
Offset 00042000: Inline data. id=003 length=010 (16) Tag length=20
Offset 00042014: CRC block with padding. Tag length=12
Offset 00042020: CRC block with padding. Tag length=992
Offset 00042400: CTZ struct: id=003 head=00000A8D (2701) size=00000FA0 (4000) Tag length=12
Offset 0004240C: CRC block with padding. Tag length=12
Offset 00042418: CRC block with padding. Tag length=1000
Offset 00042800: id=004 Creating file. Tag length=4
Offset 00042804: Regular file: id=004 name='test_003.txt'  Tag length=16
Offset 00042814: Inline data. id=004 length=000 (0) Tag length=4
Offset 00042818: CRC block with padding. Tag length=12
Offset 00042824: CRC block with padding. Tag length=988
Offset 00042C00: Inline data. id=004 length=010 (16) Tag length=20
Offset 00042C14: CRC block with padding. Tag length=12
Offset 00042C20: CRC block with padding. Tag length=992
_
(I still notice the extra empty inline data tag after opening every file though?
Is this due to the file being committed to storage to remember name and it's the only way to store file length of 0?)

Data integrity is still ok, but I do not know how much more wear this causes or other issues.
This should help OP as I think his nand may have similar capability although on 2048 byte sectors allowing 4x512 byte writes instead. For live file creation though I see the metadata_max does wonders to keep delays down so suggest creating few files in a directory is possible during time critical events.

kyrreaa, 2024-02-09:
I forgot to say @dorsadeh that any file that does not fit inline will be allocated a full eraseable block sized chunk (block_size) in the CTZ structure for the file. The ability to write 4 smaller bits to a sector will not help with the current way littlefs allocates data space.

geky, 2024-02-17:
> Here's some output from my own analyser (Is there one in littlefs, I could not find one?)

There are a couple scripts used during development:
- https://github.com/littlefs-project/littlefs/blob/master/scripts/readtree.py
- https://github.com/littlefs-project/littlefs/blob/master/scripts/readmdir.py
- https://github.com/littlefs-project/littlefs/blob/master/scripts/readblock.py

``` bash
$ ./scripts/readtree.py --help
$ ./scripts/readtree.py disk 4096
```

Though no promises they work all the time. It would be nice to make these stable and move to a typed language, but it's low-priority.

> After the CTZ block I was expecting a CRC padding out to 512 or 4096 (write size) somehow, but with 4 byte tag and 1022 length I get 1026 byte chunks at a time which seems odd.

This also caused by the above mentioned tag encoding limit. CRC padding uses the same 10-bit field in the tag, so the largest amount of padding is 1022 bytes (1023 being reserved for for deleted tags).

littlefs then writes multiple CRC tags to fill out the necessary padding, which is a hack, but at least keeps littlefs functional.

> With inline enabled it seems to be no way to actually utilize the first inline tag written as it is written during creation of the file upon open, and no writes to the file has happened yet. It [...]

I can address this in https://github.com/littlefs-project/littlefs/issues/942#issuecomment-1939618808, thanks for creating an issue.

> (I still notice the extra empty inline data tag after opening every file though?
> Is this due to the file being committed to storage to remember name and it's the only way to store file length of 0?)

Yes, basically. In theory you could omit the empty inline tag, but since littlefs then writes ~1000 bytes of padding you wouldn't really be gaining anything

geky, 2024-02-17:
> I'm looking into making significant changes to how metadata is stored so this should improve in the long-term, but that will take time since these are more involved changes.

I just thought I'd add a quick update on this. I've been making progress on this in the background, and have a prototype working that should, in theory, be able to remove both the RAM requirement for inline files and tag encoding limitations.

Though it will be a bit longer before this is usable.