littlefs-project/littlefs #662: Random lfs_dir_fetchmatch() failure

shilpa-1992, 2022-04-05:
In NOR flash ,W25Q512NW the file mount , file write , file close operations are done successfully but at times when We start to execute the code with file system mount it get hits with lfs_dir_fetchmatch () failure. Later only format resolves this.

This lfs_dir_fetchmatch () failure is seen randomly. Could you please help me to understand what might be the the causes of failure and how can we identify that.

geky, 2022-04-06:
This is unfortunately not very much information to go on. On common issue I've seen is misconfigured block devices causing the wrong block to be written or erased during use.

Have you ran tests that verify the block device can read and write various blocks on disk?

geky, 2022-04-06:
lfs_dir_fetchmatch is where the CRCs on the device are verified, so this likely means that the writes are not being stored on disk correctly.

It could also be that there are caches in the block device or block device driver that need to be flushed as a part of the sync callback.

shilpa-1992, 2022-04-06:
Sorry for the short issue description.

Scenario where Issue  faced:
Case1:
LittleFS  mounted , file open , file write, file close and unmount was done.
Later another code was flashed in XIP mode and debug of that code was done.
Now the board is handed over for LittleFS test run.
This time the mount operation was failing with below error.

corrupted dir pair{0x0,0x1}
bad block at 0x0
super block 0x0 has become unwritable.

Case2:
Binary got written to flash area by mistake and mass erase was done.
After the mass erase operation the litleFS mount was not happening.
The format was done by flashing another littlefs sample code later which the issue resolved.

why was it happening what should we do in order to avoid such 'format' or such 'LFS unable to mount ' scenario.

#define LITTLEFS_START_ADDR 0x400000

#define LITTLEFS_READ_SIZE 16
/* Maximum block program size definition */
#define LITTLEFS_PROG_SIZE 256
/* Erasable block size definition */
#define LITTLEFS_BLOCK_SIZE 4096
/* Minimum block cache size definition */
#define LITTLEFS_CACHE_SIZE 256
/* Minimum lookahead buffer size definition */
#define LITTLEFS_LOOKAHEAD_SIZE 16
 
.read = lfs_mflash_read,
  .prog = lfs_mflash_prog,
  .erase = lfs_mflash_erase,
  .sync = lfs_mflash_sync,
  .read_size = LITTLEFS_READ_SIZE,
  .prog_size = LITTLEFS_PROG_SIZE,
  .block_size = LITTLEFS_BLOCK_SIZE,
  .block_count = 1024,
  .block_cycles = 100,
  .cache_size = LITTLEFS_CACHE_SIZE,
  .lookahead_size = LITTLEFS_LOOKAHEAD_SIZE

#define LITTLEFS_START_ADDR 0x400000 : Does this mean that LittleFS area starts after 4MB. Does it make sense to change the start address to higher value so that to avoid unintended write operation.