littlefs-project/littlefs #668: Test for already opened file

SadE54, 2022-04-10:
Hi , what is the best way to know if a file is already opened ? 🤔

geky, 2022-04-10:
Hi @SadE54, littlefs currently doesn't have a mechanism for this.

If you open the same file twice, littlefs will happily create two open `lfs_file_t` structs, and the data on disk will reflect whichever file is synced or closed last.

SadE54, 2022-04-14:
I'm using static buffer for file , and I just have to work with only one file at a time bu from different tasks, that why I need to knowif the file was already opened.
I tried some things but it seems to , that if you're trying to open a non existing file , no error is returned , the file can be opened normally  😖