littlefs-project/littlefs #672: Will littfs support compression in the future？

xukaihub, 2022-04-21:


geky, 2022-04-21:
Interesting you ask this because I was looking into this quite recently.

In the general case, I don't think so. Compression requires significant (for a microcontroller) RAM, and this is sort of a dead end for littlefs. Though if you happen to come across any constant-RAM compression algorithms let me know.

Compression lets you trade off some RAM and processing power for lower disk usage, but for littlefs we usually want the tradeoff to go the other way. Flash storage is cheap, so if you can use more storage to decrease RAM consumption/performance, you can get a smaller/cheaper MCU.

---

That being said, I think it would be really interesting to look at pre-compressed littlefs images, That is, allowing you to compress littlefs offline, such as at compile-time, and then mount/decompress littlefs at run-time.

I think this would be very useful for read-only images such as those used in bootloaders, and focusing on decompression would be a much easier problem.

felipeLeast, 2022-04-26:
@geky Take a look at lz4 compression, it has a constant RAM compression, and high performance for microcontrollers. Mbed has an example in their mbed cloud project:

https://github.com/PelionIoT/mbed-cloud-client/blob/master/update-client-hub/delta-tool-internal/source/lz4.c

BenBE, 2022-04-27:
Another possible implementation for compression could be on the route of [heatshrink](https://github.com/atomicobject/heatshrink), which was developed with low/constant memory consumption in mind; n particular for low-end MCUs. Compression somewhat depends on how much RAM you allocate for it, but with even low settings the used LZS implementation works quite well.

Remains the question: Which parts of LFS could best be compressed. While there's the possibility to compress simple images like squashfs does, there's also the possibility to allow for compression of attributes/metadata or inline file contents. Both with metadata and inline file contents things should be accessible with the usual API.

Also given you mentioned low RAM consumption: You thus likely should avoid compressing the actual directory structure; especially as this might cause quite a bit of additional processing required to find information when (compressed) directories have to be scanned repeatedly for information.

geky, 2024-09-20:
Circling back on this a bit, there are some upcoming changes to the file structure that I think make compression quite a bit easier to implement, so it would be interesting to revisit this in the future.

> @geky Take a look at lz4 compression, it has a constant RAM compression

> Another possible implementation for compression could be on the route of [heatshrink](https://github.com/atomicobject/heatshrink)

Thanks @felipeLeast, @BenBE. I made a mistake in thinking there were no constant-RAM compression algorithms. Though my understanding is there is a performance cliff where having a too small amount of RAM ends up compressing poorly. That combined with needing to match encoder/decoder dictionaries creates some annoying questions around how much RAM you need for portability.

I think the most complicated part will be how the compression configuration (which algo, dict size, etc) is stored. Simply storing the configuration per-file and failing to read when an algo/RAM is not available is one option.

---

> Also given you mentioned low RAM consumption: You thus likely should avoid compressing the actual directory structure; especially as this might cause quite a bit of additional processing required to find information when (compressed) directories have to be scanned repeatedly for information.

Hmmm, I've only been thinking of compressing file data, which is a much simpler problem. Compressing metadata would be quite a bit more complicated, though not impossible... Especially since metadata is all append-only logs... 🤔

This paragraph could probably be an entire research project.

> Remains the question: Which parts of LFS could best be compressed. While there's the possibility to compress simple images like squashfs does, there's also the possibility to allow for compression [...]

Inline data is an interesting question, and will probably take some experimentation. But if you revert to uncompressed on failed compression I don't see a reason to not try compressing inline data. We usually think of inlined data as "small", but if you have a >=1 MiB block size, your inlined data is probably not "small".

For attributes it's a bit more questionable. My gut says the portability issues would not be worth the compression gains, especially since attributes are usually quite small, but my gut has been wrong before...

---

Interesting things to think about.

geky, 2025-06-01:
I've been thinking about transparent compression again recently, in the context of v3 (https://github.com/littlefs-project/littlefs/pull/1111). Compression is still out-of-scope, but now would be the time to make any carve-outs to the on-disk format if necessary.

I'm only focusing on data compression. Metadata compression would be a much more different problem. And likely low-value given that most metadata is high-entropy/relatively small (but I can always be wrong!).

One nice thing in v3 is that the physical block/fragment size can be different from the "weight" stored in the file's B-tree. This allows for file holes, but may also make introducing compression relatively easy.

---

The remaining question is how to integrate transparent compression with the least impact to the on-disk format. I think I have a pretty good idea here, so wanted to write it down:

1. Instead of a separate file type, compressed files are `LFS_TYPE_REG`, but with a standard "compressed" custom attribute.

   This custom attribute would both indicate the file is compressed, and store the compression type and config (window size, etc).

2. It makes the most sense to implement compression _in_ littlefs itself. It's invasive, but this way we could align compression to physical blocks/fragments, allowing for random reads/writes.

   At least compression can still be an opt-in ifdef feature, and potentially user controlled via callbacks in `struct lfs_config`.

The benefit of using a custom attribute instead of separate file type is that you would still be able to interact with compressed files with a littlefs driver that doesn't support compression. Reads would return garbage, but at least you would be able to `lfs_rename`, `lfs_remove`, etc, without needing to worry about breaking things.

BenBE, 2025-06-02:
If you go the route of the custom attribute (BTW, that's what NTFS does too, though parameters hard-coded) you could even allow for callbacks/hooks to be placed that replace/supplement the built-in functions. That way people could even mix different algorithms like bzip2 for their log files while using gzip for config data and heatshrink for "upgrade images" …

Another place I just imagined could be useful to have compression for is with CoW compacting/defragmentation, i.e. when you notice a directory or other on-disk structure gets above a certain threshold while you have to move it anyways you could place it in the new spot while compressing the (cleaned-up) data. Think of a large directory with lots of files that you clean up old revisions from and instead of just removing unused old revisions of attributes, you compress the compacted version of the metadata on-the-fly.

geky, 2025-06-02:
Good point! This can easily be configured on a per-file basis.

In a perfect world I'd like for all compression to be callback driven, as it avoids needing to decide the best default, and as you point out is likely data/pattern dependent.

Though we could add opt-in ifdef hooks for popular libraries.

> Another place I just imagined could be useful to have compression for is with CoW compacting/defragmentation, i.e. when you notice a directory or other on-disk structure gets above a certain [...]

The structure of littlefs means we're unlikely to implement metadata-level compression. For similar reasons we are unlikely to support filesystem-wide CoW/snapshots. (everything CoW/snapshots/compressions is at the file-level) But I suppose you could always mount a littlefs image _inside_ a littlefs file 🤔