littlefs-project/littlefs #676: LittleFS uses half of SPI Flash on STM32

muratdemirtas, 2022-04-27:
Hi,
I really appreciate that library. I ported this library to our electronic product that uses an STM32 controller and working perfectly.  
We using w25q80 8Mbit(1MB) SPI NOR flash memory for our log data. We use the append method for the log data on the text file using LittleFS. But I have faced an issue that the log file size does not exceed 512 kb while appending to a text file. We need to use all of the SPI flash memory.

I believe my configuration file is wrong because I couldn't find the right parameters for our SPI flash memory chip. I leaving my existing configuration below.

const struct lfs_config cfg =
{
	.read  = lfs_deskio_read,
	.prog  = lfs_deskio_prog,
	.erase = lfs_deskio_erase,
	.sync  = lfs_deskio_sync,
	.read_size = 16,
	.prog_size = 16,
	.block_size = 4096,
	.block_count = 256,
	.cache_size = 16,
	.lookahead_size = 16,
	.block_cycles = 500,
	.read_buffer = read_buffer,
	.prog_buffer = prog_buffer,
	.lookahead_buffer = lookahead_buffer,
};

Thanks for any helpful comment.

alexandrehagihara, 2022-04-29:
Hi, I see no problem with your config struct. I've tested it with my board and I could create a huge file with more than 512kB.
Are you sure you are respecting your nor flash specifications? For example, erasing and programming timings (or checking busy flag), page programming boundaries, block protection, etc.?

muratdemirtas, 2022-04-29:
Thx you i will check.

> Hi, I see no problem with your config struct. I've tested it with my board and I could create a huge file with more than 512kB. Are you sure you are respecting your nor flash specifications? For [...]

muratdemirtas, 2022-04-30:
I added a binary semaphore to our file operations and the problem is solved. Thanks.

closed 2022-04-30 by muratdemirtas

vtjballeng, 2024-06-02:
@muratdemirtas, which STM32 uC did you use and is your implementation public or available?