littlefs-project/littlefs #681: How can I know if a file exists?

ssergibionitlabs, 2022-05-19:
Hi everyone.
How can I tell if a file exists?
Is there a specific command?
Thanks

ssergibionitlabs, 2022-05-19:
> You can use lfs_stat()

Hi alexandrehagihara,
thank you for your answer.
I already use the lfs_stat function.
But in my opinion this function creates problems for me. I use it like this:

if (lfs_stat(&lfs, CONFIG_EMG0_FILENAME, &info) != LFS_ERR_OK)
         SaveEMGSettings(&EMGSettings, 0);

That is, if the file does not exist then I create it with the SaveEMGSettings function.

Everything works fine. Sometimes, however, the microcontroller crashes.
It seems that the microcontroller ends up in the wrong memory location.
It returns me the following error message: "unknown function at 0x....."
I use a microcontroller from the nRF52 family and SEGGER Embedded Studio for ARM.

If I remove all lfs_stat functions everything works fine.
Maybe I misuse the function?
Can you help me?
Thanks.

alexandrehagihara, 2022-05-19:
Are you running code from RAM? It seems like some stack overflow problem.
Who is popping this error message? There is nothing like that in lfs code, it must be something related to your port or framework.

ssergibionitlabs, 2022-05-19:
> Are you running code from RAM? It seems like some stack overflow problem. Who is popping this error message? There is nothing like that in lfs code, it must be something related to your port or framework.

The error message is returned by the compiler.
littlefs gives me no errors.
I just know that if I remove that instruction the microcontroller won't crash.
I load the firmware on the microcontroller flash.