littlefs-project/littlefs #684: Compiler warnings in `lfs_dir_traverse` about `dir = dir` etc

dpgeorge, 2022-05-23:
Removal of recursion in `lfs_dir_traverse` in 8109f282668d0396c1a8df05d38ed215b0b8feac added 4 lines which give warnings on some compilers (eg clang):
```
               dir = dir;
               off = off;
               ptag = ptag;
               attrs = attrs;
```
I suggest to remove these 4 lines.

dpgeorge, 2022-05-23:
Example of warning:
```
../../lib/littlefs/lfs2.c:868:27: error: explicitly assigning value of variable of type 'int' to itself [-Werror,-Wself-assign]
                attrcount = attrcount;
                ~~~~~~~~~ ^ ~~~~~~~~~
```

geky, 2022-05-23:
Thanks for creating an issue, it sounds like we should add clang to CI with -Werror like we have GCC.

I'll look at these for the next patch release.

dpgeorge, 2022-05-23:
> it sounds like we should add clang to CI with -Werror like we have GCC

Yes I think that would be a good idea, to prevent other warnings from creeping in.

So, is it the correct fix to just remove these 5 lines?  (We need to apply that patch to our copy of littlefs.)

geky, 2022-05-23:
> So, is it the correct fix to just remove these 5 lines? (We need to apply that patch to our copy of littlefs.)

Yes those lines have no effect. I think I left them as a documentation tool since those variables are the "stack frame" of the lfs_dir_traverse function.

geky, 2022-09-07:
Sorry for about the timing. I have a PR for this up here now: https://github.com/littlefs-project/littlefs/pull/724