littlefs-project/littlefs #690: Tips for reducing/eliminating wear when using LittleFS?

embeddedpenguin, 2022-05-31:
I've been playing with the simulator and am impressed with the potential results vs FatFS. However, on some devices that are meant to be deployed for years at a time, a file system failure could mean total device failure. Is there some way to reset the filesystem (kind of like defragging it) to eliminate the wear on the device? Or would the only way to do this be moving all contents off the chip -> chip erase -> recreating the littlefs?

AdvelC, 2022-06-07:
This simulator ([http://littlefs.geky.net/demo.html](url)?
In #660 it was mentioned "_Ah, that simulation is unfortunately quite out of date. It's still using v1, which relied on bad-block reporting, but didn't provide true dynamic wear-leveling. v2 now has dynamic wear-leveling (which introduced the block_cycles configuration option.)_"

embeddedpenguin, 2022-06-09:
Yeah I was using that simulator. I see it is out of date. I'll look more into the new dynamic wear-leveling. Thanks

closed 2022-07-19 by embeddedpenguin