littlefs-project/littlefs #694: Feature request: An option to use LittleFS with no directories

embeddedpenguin, 2022-06-04:
It's not really a LFS issue. I am just on a microcontroller with very little memory, and I think eliminating the directory code would save on code size by a lot. I would like an option to use littlefs for file operations only. My use case is the following:
I have a few files I need to make a lot of reads and writes to. Since there names will always be known and I have no need of a tree structure to hold files.

I actually tried adding my own LFS_NO_DIR and ifndef guarding the directory code out, but some file functions rely on directory functions currently so I bailed on that idea for now. This request may seem kind of unorthodox (from what I know about tiny filesystems), but I think it makes a lot of sense for systems with low low low memory requirements. I'd love to know what you think about it!

geky, 2022-09-07:
This is a good idea, and worth exploring. However I suspect it would provide a surprisingly low amount of savings.

You always need at least the root directory, since you need to keep track of files somewhere. And if the root directory is larger than a single block, littlefs stores the root directory in multiple blocks, creating a sort of tree structure internal to the directory. The extra cost to let the directory store references to other directories is not that much.

Most of the more complicated directory functions, `lfs_mkdir/lfs_dir_read/etc` are standalone functions that should be dropped at link time if you're compiling with `--gc-sections`.

embeddedpenguin, 2023-09-16:
Closing this because its been a while. If I ever revisit this idea, I'll reopen with a pr

closed 2023-09-16 by embeddedpenguin