littlefs-project/littlefs #701: File append periodically stuck in lfs_ctz_traverse() while loop

klemdev, 2022-06-08:
Hi.
I have a question about some strange behavior of appending data to a file.
I have an 16GB SD card connected to STMF405 and a simple program appending 511 bytes to a file in an infinite loop.

It looks like every 2033rd or 2034th cycle append is stuck for some time, and it seems that larger the file longer it is stuck. All iterations in between append data smoothly.
At around 200.000 iterations timeout can be few minutes!!!

Here is my config:
`
.read_size = 512,
.prog_size = 512,
.block_size = 512,
.block_count = 31207424,
.block_cycles = 500,

.cache_size = 512,
.lookahead_size = 512,
.read_buffer      = lfs_read_buf,
.prog_buffer      = lfs_prog_buf,
.lookahead_buffer = lfs_lookahead_buf,
.file_max = LFS_FILE_MAX,
`
where read, prog and lookahead buffers are all 512 byte arrays.

Is there something wrong with my config?

Here is a bit of trace in case when it get stuck. It looks like that "while (true)" loop in lfs_ctz_traverse() is taking long to finish:

`
trace: lfs_file_write(0x200012f8, 0x20001370, 0x2001fdd8, 511)
lfs_file_rawwrite() call lfs_file_flushedwrite()
lfs_file_flushedwrite() file->pos+nsize=5258190 file->ctz.size=5257679 file->flags=2307
lfs_file_flushedwrite() lfs->cfg->cache_size=512 lfs->cfg->metadata_max=0 lfs->cfg->block_size=512
lfs_file_flushedwrite() call lfs_ctz_extend()
lfs_ctz_extend() call lfs_alloc()
lfs_alloc() found free block
lfs_alloc() return 0
lfs_file_flushedwrite() call lfs_bd_prog()
lfs_file_flushedwrite() call lfs_alloc_ack()
lfs_file_flushedwrite() call lfs_ctz_extend()
lfs_ctz_extend() call lfs_alloc()
lfs_alloc() call lfs_fs_rawtraverse()
lfs_fs_rawtraverse() call lfs_ctz_traverse()
lfs_ctz_traverse() start while (true) index=10431 size=5257679

!!!!!!!!!! LONG DELAY STARTS HERE !!!!!!!!!!
lfs_ctz_traverse() end while (true) index=10430 count=1
lfs_ctz_traverse() end while (true) index=10428 count=2
lfs_ctz_traverse() end while (true) index=10426 count=2
...
!!!!!!!!!! LONG DELAY ENDS HERE !!!!!!!!!!

lfs_ctz_traverse() return index == 0.
lfs_fs_rawtraverse() call lfs_ctz_traverse()
lfs_ctz_traverse() start while (true) index=10431 size=5257772

!!!!!!!!!! LONG DELAY HERE !!!!!!!!!!
lfs_ctz_traverse() end while (true) index=10430 count=1
lfs_ctz_traverse() end while (true) index=10428 count=2
lfs_ctz_traverse() end while (true) index=10426 count=2
...
!!!!!!!!!! LONG DELAY ENDS HERE !!!!!!!!!!

lfs_ctz_traverse() return index == 0.
lfs_alloc() found free block
lfs_alloc() return 0
lfs_file_flushedwrite() call lfs_bd_prog()
lfs_file_flushedwrite() call lfs_alloc_ack()
trace: lfs_file_write -> 511
trace: lfs_file_close(0x200012f8, 0x20001370)
trace: lfs_file_close -> 0
trace: lfs_file_opencfg(0x200012f8, 0x20001370, "test_200ms_511b.txt", 903, 0x200010ec {.buffer=0x200010f8, .attrs=0x0, .attr_count=0})
trace: lfs_file_opencfg -> 0
trace: lfs_file_seek(0x200012f8, 0x20001370, 0, 2)
trace: lfs_file_seek -> 5258190
`

Any comment is much appreciated.
I can provide any additional debugs if required by someone familiar with under the hood implementation.

klemdev, 2022-06-09:
Hi.

I forgot to mention that SD card is connected to the MCU board via the SPI. I know this is not the best solution but anyway, would such delays be normal and why only 2033th or 2034th append takes so long?

I also tried by restarting the board and continue to append to a ~59MB file and it takes 23 minutes before it does the first append, and all following 2032 appends run smoothly.
Does littlefs read the whole file first?

I am really puzzled what this 2033/2034 has to do with this issue.

kaetemi, 2022-07-05:
Traverse will read the first bytes of each block in each file back to front to detect the allocated blocks. So, if your read size equals your block size, it will effectively read the entire file. Use bigger blocks if you cannot make smaller reads.

Your magic number that triggers the traversal is the `lookahead_size`.

klemdev, 2022-07-08:
@kaetemi thank you very much for your reply.

Since I am using littlefs on an SD card, my block size is fixed to 512. I think I can do any kind of read, but should write/program be fixed to 512, too?
As I see I have to stick to rules:

assert(cache_size % read_size == 0);
assert(cache_size % prog_size == 0);
assert(block_size % cache_size == 0);

This gives that they all have to be 512 as I have now :-(
Am I right here or program size can be lower?

If you have any experience with SD cards, what do you think would be proper config?

Thank you very much for your help.

kaetemi, 2022-07-08:
Just use a bigger block size that's a multiple of 512, and recalculate appropriately in your implementation callbacks to remap multiple hardware blocks to one lfs block.

muratdemirtas, 2022-07-18:
@klemdev https://github.com/littlefs-project/littlefs/issues/712

sslupsky, 2022-07-18:
> Just use a bigger block size that's a multiple of 512, and recalculate appropriately in your implementation callbacks to remap multiple hardware blocks to one lfs block.

I think that might be problematic for some.  If you are using a device with a large block size, say an SPI NAND with a 4096 block, lfs requires 3 x 4096 bytes of memory for the read, prog and cache buffers.  So, multiplying those numbers "breaks the bank" so to speak on ram requirements for resource constrained mcu's.

kaetemi, 2022-07-18:
@sslupsky Nope. Your buffers only need to be sized to fit your minimum read and write sizes. Your block size can be as large as you like, and should be a pow2 multiple of those.

sslupsky, 2022-07-19:
@kaetemi apologies, I misunderstood what you were referring to.  You are referring to the erase block size?

```
 // Size of an erasable block. This does not impact ram consumption and
    // may be larger than the physical erase size. However, non-inlined files
    // take up at minimum one block. Must be a multiple of the read
    // and program sizes.
    lfs_size_t block_size;
```

Why would you want to increase the block size?  In my case, block size is 128KB which is 64x the size of the prog buffer size.

kaetemi, 2022-07-19:
@sslupsky Yes. The OP is using a tiny block size that equals his minimum read size. He should increase his block size, otherwise any traversal will be completely reading every single file. Your block size should be several multiples of your minimum *read* size.

The lookahead buffer contains one bit per block, marking for a select region which blocks are allocated already. His file is blocking every time he reaches the end of the lookahead buffer, and it's blocking bad because his configuration is reading every byte on his filesystem just to refresh the lookahead.

And you should really set your minimum read size to the minimum.

> @kaetemi <https://github.com/kaetemi> apologies, I misunderstood what you were referring to. You are referring to the erase block size?
>
> ` // Size of an erasable block. This does not impact ram [...]
`

sslupsky, 2022-07-19:
>  it's blocking bad because his configuration is reading every byte on his filesystem just to refresh the lookahead. 

@kaetemi Ahh, it that what traversal does?  And traversal only reads the minimum read size?  So, if the read size is 1 byte then traversal will only have to read one byte per block during the traversal?

Thanks for the info ... this is helpful.

kaetemi, 2022-07-19:
It'll read the first 8 bytes in each block of each file, yes. It's just a linked chain of block pointers from back to front for each file. That's how it tracks which blocks are in use.

For actual reads, it can read directly to whatever buffer you pass to lfs read function, and it'll happily read out a whole block at once that way. The read buffer is really just for those 8 bytes and for buffering very tiny reads.

klemdev, 2022-08-18:
@kaetemi Thank you very much for your comments and explanation.
So, if I understand correctly, to increase block size from 512 to 4096 (8 times increase) I did:
1. in lfs_config:
// Modify actual nomber of blocks by dividing SD card blocks by 8:
```
.block_count = <number of SD card 512B blocks> / 8
.block_size = 4096
.lookahead_size = 4096
.lookahead_buffer = lfs_lookahead_buf
```

and increased static lookahead buffer, like:
`uint8_t lfs_lookahead_buf[4096]`

2. My user provided functions just call SPI transfer function, like:
```
sd->Read(buffer, (uint64_t)(block * c->block_size + off), size);
sd->Program(buffer, (uint64_t)block * c->block_size + off, size);
```

This usually works, but I think there are still issues. Sometimes it works smoothly, but a bit slower. Sometimes it is just stuck at 2nd write.

Am I still missing something in changing this to a bigger block?

Thank you very much for any comment. Really appreciated.