littlefs-project/littlefs #712: lfs_file_seek hangs.

muratdemirtas, 2022-07-18:
`[02:52:10:236] ./main/bab.module.littlefs/littlefs/lfs.c:5599:trace: lfs_file_seek(0x3ffc98f8, 0x3ffdcaa0, 802304, 0)␍␊
[02:52:10:236] ./main/bab.module.littlefs/littlefs/lfs.c:5604:trace: lfs_file_seek -> 802304␍␊
[02:52:10:236] ./main/bab.module.littlefs/littlefs/lfs.c:5563:trace: lfs_file_read(0x3ffc98f8, 0x3ffdcaa0, 0x3ffdce3c, 128)␍␊
[02:52:10:280] ./main/bab.module.littlefs/littlefs/lfs.c:5568:trace: lfs_file_read -> 128␍␊
[02:52:10:280] ./main/bab.module.littlefs/littlefs/lfs.c:5563:trace: lfs_file_read(0x3ffc98f8, 0x3ffdcaa0, 0x3ffdce3c, 128)␍␊
[02:52:10:280] ./main/bab.module.littlefs/littlefs/lfs.c:5568:trace: lfs_file_read -> 128␍␊
[02:52:10:280] ./main/bab.module.littlefs/littlefs/lfs.c:5563:trace: lfs_file_read(0x3ffc98f8, 0x3ffdcaa0, 0x3ffdce3c, 128)␍␊
[02:52:10:280] ./main/bab.module.littlefs/littlefs/lfs.c:5568:trace: lfs_file_read -> 128␍␊
[02:52:10:280] ./main/bab.module.littlefs/littlefs/lfs.c:5563:trace: lfs_file_read(0x3ffc98f8, 0x3ffdcaa0, 0x3ffdce3c, 128)␍␊
[02:52:10:323] ./main/bab.module.littlefs/littlefs/lfs.c:5568:trace: lfs_file_read -> 128␍␊
[02:52:10:323] esp32_Read: 3r /littlefs/JanStore.db 512 512 OK␍␊
[02:52:10:323] esp32_Read: 1r /littlefs/JanStore.db 512 801792[801792] ␍␊
[02:52:10:323] ./main/bab.module.littlefs/littlefs/lfs.c:5599:trace: lfs_file_seek(0x3ffc98f8, 0x3ffdcaa0, 801792, 0)␍␊
[02:52:10:323] ./main/bab.module.littlefs/littlefs/lfs.c:5604:trace: lfs_file_seek -> 801792␍␊
[02:52:10:323] ./main/bab.module.littlefs/littlefs/lfs.c:5563:trace: lfs_file_read(0x3ffc98f8, 0x3ffdcaa0, 0x3ffdce3c, 128)␍␊
[02:52:10:389] ./main/bab.module.littlefs/littlefs/lfs.c:5568:trace: lfs_file_read -> 128␍␊
[02:52:10:389] ./main/bab.module.littlefs/littlefs/lfs.c:5563:trace: lfs_file_read(0x3ffc98f8, 0x3ffdcaa0, 0x3ffdce3c, 128)␍␊
[02:52:10:389] ./main/bab.module.littlefs/littlefs/lfs.c:5568:trace: lfs_file_read -> 128␍␊
[02:52:10:389] ./main/bab.module.littlefs/littlefs/lfs.c:5563:trace: lfs_file_read(0x3ffc98f8, 0x3ffdcaa0, 0x3ffdce3c, 128)␍␊
[02:52:10:389] ./main/bab.module.littlefs/littlefs/lfs.c:5568:trace: lfs_file_read -> 128␍␊
[02:52:10:389] ./main/bab.module.littlefs/littlefs/lfs.c:5563:trace: lfs_file_read(0x3ffc98f8, 0x3ffdcaa0, 0x3ffdce3c, 128)␍␊
[02:52:10:432] ./main/bab.module.littlefs/littlefs/lfs.c:5568:trace: lfs_file_read -> 128␍␊
[02:52:10:432] esp32_Read: 3r /littlefs/JanStore.db 512 512 OK␍␊
[02:52:10:518] esp32_Write: 1w /littlefs/JanStore.db 512 801792[801792] ␍␊
[02:52:10:566] ./main/bab.module.littlefs/littlefs/lfs.c:5599:trace: lfs_file_seek(0x3ffc98f8, 0x3ffdcaa0, 801792, 0)␍␊
[02:52:10:566] anan 1 ␍␊
[02:52:10:566] anan 33 ␍␊
[02:52:10:566] anan 2 ␍␊
[02:52:10:566] anan 2333 ␍␊
[02:52:10:566] ./main/bab.module.littlefs/littlefs/lfs.c:5604:trace: lfs_file_seek -> 801792␍␊
[02:52:10:566] write size 512, file /JanStore.db,  cache size 4./main/bab.module.littlefs/littlefs/lfs.c:5581:trace: lfs_file_write(0x3ffc98f8, 0x3ffdcaa0, 0x3ffe2c20, 512)␍␊
[02:52:13:423] ./main/bab.module.littlefs/littlefs/lfs.c:5586:trace: lfs_file_write -> 512␍␊
[02:52:13:423] esp32_Write: 3w /littlefs/JanStore.db OK␍␊
[02:52:13:423] esp32_Read: 1r /littlefs/JanStore.db 512 807424[807424] ␍␊
[02:52:13:423] ./main/bab.module.littlefs/littlefs/lfs.c:5599:trace: lfs_file_seek(0x3ffc98f8, 0x3ffdcaa0, 807424, 0)␍␊
[02:52:13:423]  file pos : 802304  file ctz size: 2958336␍␊
[02:52:13:423]  file pos : 802305 file ctz size:2958336␍␊
[02:52:13:423]  file pos : 802306 file ctz size:2958336␍␊`

muratdemirtas, 2022-07-18:
@ #591

sslupsky, 2022-07-18:
How big is the file?  I have found that as the file size grows, lfs can take a very long time to seek to the end of the file.

muratdemirtas, 2022-07-18:
> How big is the file? I have found that as the file size grows, lfs can take a very long time to seek to the end of the file.

JanStore.db file is 2958336 bytes

and i want to write a 100 byte.

alexandrehagihara, 2022-07-18:
If you are using sqlite, you aren't just appending 100 bytes to the end of your database file. You could be relocating several blocks just for inserting a single record.

muratdemirtas, 2022-07-18:
> If you are using sqlite, you aren't just appending 100 bytes to the end of your database file. You could be relocating several blocks just for inserting a single record.

is this littlefs library is suitable for sql files?

sslupsky, 2022-07-18:
> JanStore.db file is 2958336 bytes, 3MB
> 
> and i want to write a 100 byte.

I believe that is a big file size for littlefs.  Are you using serial flash?  Opening the file and seeking the end of file could take several minutes.  I think I read in another thread where someone mentioned the time it took to append data to a 59MB log file was about 23 minutes.  I think once the file is opened and the seek is done, subsequent writes append the data quickly.  So, for the append use case I think it is best that you open the file and leave it open and keep writing to the file.

> is this littlefs library suitable for work with sql files?

I am not certain but I think for a large db, you would have to be prepared for worst case which would be similar to seeking the end of file each time.

muratdemirtas, 2022-07-18:
> I believe that is a big file size for littlefs. Are you using serial flash? Opening the file and seeking the end of file could take several minutes. I think I read in another thread where someone [...]

thanks a lot of. I'm using 16GB Type: SDHC/SDXC␍ sd card. I decided to stay with FatFS.

sslupsky, 2022-07-18:
I believe the journaling nature of the file system causes this behaviour.  It provides the power resilience that FAT does not have but there is a significant performance cost.

muratdemirtas, 2022-07-18:
> I believe the journaling nature of the file system causes this behaviour. It provides the power resilience that FAT does not have but there is a significant performance cost.

Looks like we have to design the hardware systems which detects a power loss and provide 3-4 seconds to supply esp32 voltage before brownout.

kaetemi, 2022-07-19:
Same comment as in #701, if your minimum read size is the same as your block size, all file system traversals will read every single file in its entirety.

You may want to increase `--block_size=512` to 4k, 32k, or even bigger, depending on your requirements.

On mount or any traversal, **read size** * **block count** is the amount of bytes that will be read!

sslupsky, 2022-07-19:
> You may want to increase `--block_size=512` to 4k, 32k, or even bigger, depending on your requirements.
> 
> On mount or any traversal, **read size** * **block count** is the amount of bytes that will be read when the filesystem is full!

SD cards use nand flash memory which have erase block sizes reaching into the MB's.  So, your block erase size is likely much larger than 512 bytes.  However, it does not appear this parameter is well documented for SD cards.  I looked up the block size for a 16GB SD card I have plugged into a RPI that shows a block erase size of 4MB.

klemdev, 2022-11-24:
@sslupsky 
So, for the append use case I think it is best that you open the file and leave it open and keep writing to the file.

But if you cut power, wouldn't you loose complete file?

geky, 2022-12-02:
> I believe the journaling nature of the file system causes this behaviour. It provides the power resilience that FAT does not have but there is a significant performance cost.

Just to clarify this isn't an inherent issue with journaling, it happens because of a poor design choice in in the file data-structure. Something more B-tree like can avoid this, with some complexity trade-offs. I have some plans to address this but the implementation will take some time.

The first issue for this was here: https://github.com/littlefs-project/littlefs/issues/27 (commenting for tracking purposes)

> On mount or any traversal, read size * block count is the amount of bytes that will be read when the filesystem is full!

Maybe interesting to note this will also be improved with B-trees. They move the tree metadata out of the data blocks, so even with a large read_size you only need to read the inner nodes of the tree to traverse.

> So, for the append use case I think it is best that you open the file and leave it open and keep writing to the file.

If you append, no matter how you append, it will be orders of magnitude faster than random writes. If you're seeing performance issues with appends that may be worth a different issue.

> But if you cut power, wouldn't you loose complete file?

Calling `lfs_file_sync` at good checkpoints will avoid this, but it does have a cost with needing to copy the last block due to padding issues.

---

Also sorry for inconsistent replies to open issues.

closed 2024-03-15 by muratdemirtas