littlefs-project/littlefs #713: Compile errors with VisualStudio

Dave-Nocturnal, 2022-07-19:
I am trying to compile lfs with Visual Studio to support unit testing and debug.   I get some rather obvious compiler errors.  Am I missing something?  How do other compilers accept this?  Here is what I find:

```
    // next step, clean up orphans
    err = lfs_fs_preporphans(lfs, -hasparent);
```
The variable ```hasparent``` is declared as a local boolean.   How can you make a boolean negative?   This is a warning.

```
    info->type = lfs_tag_type3(tag);
```
The variable ```tag``` is an int32_t and this treats it as a in8_t.  This is a warning

```
    return lfs_npw2((a & -a) + 1) - 1;
```
The variable ```a``` is passed in as a uint32_t.   This code attempts to take the negative of an unsigned integer.  This is a compile error.  There are two places where this same pattern occurs.

I would think other compilers will have the same kinds of issues.   Are these simply accepted as part of lfs?  This is my first time building lfs so I am unsure of the current development state.  I have heard good things about it from others but these warnings and errors are concerning and appear to be something all compilers would face.

kaetemi, 2022-07-19:
I was wondering about these as well, but I decided to just disable the warnings on our branch. :(

Other than that, with some small changes everything works fine. I can't get the test cases to build under MSVC, though! It uses some specific C extension to deduce the type of integers...

Our changes for MSVC support (plus our feature PRs) are in the following branch.

https://github.com/BRTSG-FOSS/littlefs/commits/fork/esd

davidefer, 2022-12-05:
Would it be possible to fix these warnings?

henri62, 2026-10-29:
There are many more warnings that shall be addressed sometime. If you enable -Wconversion and -Wshadow you see a bunch more. That does not really depend on the used environment.

Especially the file lfs_util.h is annoying and I have to patch it to get it to build in my project with a lot of even more strict checks.

I have patched these:

lfs_util.h@146 : cast the return value of __builtin_clz():
    return 32 - (uint32_t)__builtin_clz(a-1);

lfs_util.h@163 : cast the return value of __builtin_ctz():
    return (uint32_t)__builtin_ctz(a);

lfs_util.h@172 : cast the return value of __builtin_popcount():
    return (uint32_t)__builtin_popcount(a);