littlefs-project/littlefs #725: Slow write on file after flush file changes

d1307, 2022-09-11:
Hi
I have a file and i use it in append mode,
After some changes on file it will be synced by a freertos task.
First write after sync is too slow when i tried to dig deeper, i realized lfs_ctz_extend is cause of it.
As i understood it copies all previous changs of file block to a new block but why it is necessary after each sync? 

It is very annoying because i using sd card and block size is 128kb :/

d1307, 2022-09-15:
Thanks for your response.

I think I have to deal with this for a while 🥲😅

stokdam, 2023-01-24:
One cause may be also the fact that copy is performed one byte at a time. In my case (1 MB erase blocks), this causes a massive performance loss, since this kind of implementation cannot take advantage of optimized memcpy while copying data into cache.

            // just copy out the last block if it is incomplete
            if (noff != lfs->cfg->block_size) {
                for (lfs_off_t i = 0; i < noff; i++) {
                    uint8_t data;
                    err = lfs_bd_read(lfs,
                            NULL, rcache, noff-i,
                            head, i, &data, 1);
                    if (err) {
                        return err;
                    }

                    err = lfs_bd_prog(lfs,
                            pcache, rcache, true,
                            nblock, i, &data, 1);
                    if (err) {
                        if (err == LFS_ERR_CORRUPT) {
                            goto relocate;
                        }
                        return err;
                    }
                }`

A possible improvement may be to perform the copy by large enough chunks of byte. This would require to define an additional buffer in order to keep data between the read from the previous block and write to the new one.