littlefs-project/littlefs #731: Bad block at 0x0

tomvarghese, 2022-09-29:
**The problem:   When I try to write data to a file, it fails.**

Here is the debug output I'm seeing:

```
../Core/Lib/LittleFS/lfs.c:5338:trace: lfs_mount(0x24001834, 0x8019cbc {.context=0, .read=0x800851d, .prog=0x8008549, .erase=0x8008575, .sync=0x8008599, .read_size=256, .prog_size=256, .block_size=4096, .block_count=16384, .block_cycles=100000, .cache_siBSP_QSPI_Read(0x24001900, 0, 256)
BSP_QSPI_Read(0x24001900, 4096, 256)
BSP_QSPI_Read(0x24001900, 4096, 256)
BSP_QSPI_Read(0x24001900, 4096, 256)
../Core/Lib/LittleFS/lfs.c:5357:trace: lfs_mount -> 0
Filesystem mounted successfully
Button state changed, new state = 0x7f
../Core/Lib/LittleFS/lfs.c:5478:trace: lfs_file_open(0x24001834, 0x240018ac, "SiriusCpu.reg", 103)
BSP_QSPI_Read(0x24001900, 0, 256)
BSP_QSPI_Read(0x24001900, 4096, 256)
BSP_QSPI_Read(0x24001900, 4096, 256)
BSP_QSPI_Read(0x24001900, 4096, 256)
BSP_QSPI_Write(0x24001a00, 4096, 256)
BSP_QSPI_Read(0x24001900, 4096, 256)
BSP_QSPI_Read(0x24001900, 4096, 256)
BSP_QSPI_Erase_Block(0)
BSP_QSPI_Read(0x24001900, 0, 256)
BSP_QSPI_Write(0x24001a00, 0, 256)
BSP_QSPI_Read(0x24001900, 0, 256)
../Core/Lib/LittleFS/lfs.c:1886:debug: Bad block at 0x0
../Core/Lib/LittleFS/lfs.c:1891:warn: Superblock 0x0 has become unwritable
../Core/Lib/LittleFS/lfs.c:5484:trace: lfs_file_open -> -36

```

I'm probably doing something wrong, but I'm not quite sure where to start.  I've used littlefs about five years ago and came back to it last week with the hope of quickly implementing some wear leveled flash storage.

I have a 64MByte (512MBit) QSPI flash on an STM32H7 running baremetal code.  I'm hoping somebody can take a quick look and spot my mistake.

 I have repeatedly beat on the QSPI flash erasing and writing all of it using 4K read/writes and 256 byte read/writes.  No failures that I have found so far.

This is my configuration for littlefs:

```
#define FILE_SYSTEM_SIZE  (64 * 1024 * 1024)
#define N25Q512A_PAGE_SIZE  256
#define N25Q512A_SUBSECTOR_SIZE 4096

// variables used by the filesystem
static lfs_t lfs;
static lfs_file_t file;

uint8_t lfs_read_buf[N25Q512A_PAGE_SIZE];
uint8_t lfs_prog_buf[N25Q512A_PAGE_SIZE];
uint8_t lfs_lookahead_buf[N25Q512A_PAGE_SIZE/8];
uint8_t lfs_file_buf[N25Q512A_PAGE_SIZE];

// configuration of the filesystem is provided by this struct
const struct lfs_config cfg = {
    // block device operations
    .read  = block_device_read,
    .prog  = block_device_prog,
    .erase = block_device_erase,
    .sync  = block_device_sync,

    // block device configuration
    .read_size = N25Q512A_PAGE_SIZE,
    .prog_size = N25Q512A_PAGE_SIZE,
    .block_size = N25Q512A_SUBSECTOR_SIZE,
    .block_count = FILE_SYSTEM_SIZE / N25Q512A_SUBSECTOR_SIZE,
	.block_cycles = 100000,
    .cache_size = N25Q512A_PAGE_SIZE,
    .lookahead_size = N25Q512A_PAGE_SIZE/8,
	.read_buffer = lfs_read_buf,
	.prog_buffer = lfs_prog_buf,
	.lookahead_buffer = lfs_lookahead_buf,
};
```

Here is the code I'm running:

```
int FsInit(void)
{
	if (0 == BSP_QSPI_Init())
	{
//		TestQSpi();

		// mount the filesystem
		int err = lfs_mount(&lfs, &cfg);

		// reformat if we can't mount the filesystem
		// this should only happen on the first boot
		if (err)
		{
			DBG("Unable to detect a filesystem\n");
			if ( lfs_format(&lfs, &cfg) >= 0)
			{
				DBG("Formatted new filesystem\n");
			   if (lfs_mount(&lfs, &cfg) >= 0)
					DBG("Filesystem mounted successfully\n");
			   else
				   return -1;
			}
		}
		else
		{
			DBG("Filesystem mounted successfully\n");
		}
	}
	return 0;
}

int FsWrite(const char *name, const void *buffer, size_t nbytes)
{
	uint32_t n = 0;
	if (lfs_file_open(&lfs, &file, name, LFS_O_RDWR | LFS_O_CREAT) == 0 )
	{
	   if (lfs_file_write(&lfs, &file, buffer, nbytes) == 0) n = nbytes;
	   lfs_file_close(&lfs, &file);
	   return n;
	}
	return 0;
}

```

FsInit() is called when the system starts.  Then on demand I call FsWrite() to put 2230 bytes into a file. 

#define static const char *NvRegFileName = "SiriusCpu.reg";

int SaveCpuRegistersToFile(void)
{
	int nbytes = sizeof(Interface_t);
	if (FsWrite(NvRegFileName, (const void *) GetRegLoc(), nbytes) != nbytes)
	{
		CONSOLE("ERROR: Unable to write registers to NV memory!!!");
		return -1;
	}
	return 0;
}

I hope it is something simple and stupid.  Any help or guidance would be greatly appreciated.

Thanks in advance!

bloodmage, 2022-09-30:
Your block_cycles is set to 100000, that means any meta block (like block 0, 1 and any block for dir and attr data)
will be cycled 100000 times before lfs thinks it's tired and move to another block. 
It's better to set it to 100 and meta block will relocate after 100 erase-prog cycles, thus prevents worn out.

tomvarghese, 2022-10-05:
The NOR SPI flash is rated at 100K erase cycles.  Why would I set it to 100?

From past experience the numbers are quite conservative, I tested one to over 200K before I gave up.  Of course, this was at an office at a comfortable temperature.  May be if I pushed it to the max rating of the device it will be a different matter.

tomvarghese, 2022-10-05:
I did a full erase of the flash manually and now it gets past this point.

closed 2022-10-05 by tomvarghese