littlefs-project/littlefs #732: Bug in lfs_ctz_traverse

bloodmage, 2022-09-30:
I found a strange behavior when file size is at margin of  block size, which will confuse the traverse function and leave a block untracked. This may lead to data corruption.

Test program:

```
  #include <stdio.h>
  #include "lfs.h"
  #include <memory.h>
  struct mockbd {
      int blksize;
      int blkcnt;
      char* cleaned;
      char* proged;
      char* data;
      int* cycles;
  };
  
  #ifdef __cplusplus
  extern "C" {
  #endif
      void makebd(struct mockbd* bd, int blksize, int blkcnt);
      void feedcfg(const mockbd* bd, lfs_config* cfg);
      void dumpcycles(const mockbd* bd);
      void destorybd(struct mockbd* bd);
  #ifdef __cplusplus
  }
  #endif
  
  void makebd(struct mockbd* bd, int blksize, int blkcnt) {
      bd->blksize = blksize;
      bd->blkcnt = blkcnt;
      bd->cleaned = (char*)malloc(blksize * blkcnt);
      bd->proged = (char*)malloc(blksize * blkcnt);
      bd->data = (char*)malloc(blksize * blkcnt);
      bd->cycles = (int*)malloc(blkcnt * sizeof(int));
      assert(bd->cleaned);
      assert(bd->proged);
      assert(bd->data);
      assert(bd->cycles);
      memset(bd->cleaned, 0, blksize * blkcnt);
      memset(bd->proged, 0, blksize * blkcnt);
      memset(bd->data, 0, blksize * blkcnt);
      memset(bd->cycles, 0, blkcnt * sizeof(int));
  }
  void destorybd(struct mockbd* bd) {
      free(bd->cleaned);
      free(bd->proged);
      free(bd->data);
      free(bd->cycles);
  }
  
  static int read(const struct lfs_config* c, lfs_block_t block,
      lfs_off_t off, void* buffer, lfs_size_t size) {
      struct mockbd* bd = (struct mockbd*)c->context;
      //Check range
      assert(block >= 0 && block < bd->blkcnt);
      //Check offset
      assert(off >= 0 && off < bd->blksize);
      assert(off + size >= 0 && off + size <= bd->blksize);
      assert(buffer);
      //Check data written
  
      bool readunknown = false;
      bool readempty = false;
      int base = block * bd->blksize + off;
      for (int i = 0; i < size; i++) {
          if (!bd->cleaned[base + i] && !bd->proged[base + i]) {
              readunknown = true;
          }
          if (bd->cleaned[base + i] && !bd->proged[base + i]) {
              readempty = true;
          }
      }
  
      if (readunknown) {
          printf("Read unknown data blk %d, off %d, size %d\n", block, off, size);
      }
      if (readempty) {
          printf("Read empty data blk %d, off %d, size %d\n", block, off, size);
      }
  
      //printf("Read block %d (%d-%d)", block, off, off + size);
      memcpy(buffer, bd->data + base, size);
      return 0;
  }
  
  static int prog(const struct lfs_config* c, lfs_block_t block,
      lfs_off_t off, const void* buffer, lfs_size_t size) {
      struct mockbd* bd = (struct mockbd*)c->context;
      //Check range
      assert(block >= 0 && block < bd->blkcnt);
      //Check offset
      assert(off >= 0 && off < bd->blksize);
      assert(off + size >= 0 && off + size <= bd->blksize);
      assert(buffer);
  
      //Check requirement of erased
      bool blockempty = true;
      int base = block * bd->blksize;
      for (int i = 0; i < bd->blksize; i++) {
          if (bd->proged[base + i]) {
              blockempty = false;
          }
      }
  
      if (!blockempty) {
          printf("Trying to prog not empty blk %d, off %d, size %d\n", block, off, size);
          return LFS_ERR_IO;
      }
  
      //Change status
      for (int i = 0; i < bd->blksize; i++) {
          bd->cleaned[base + i] = false;
          bd->proged[base + i] = true;
      }
      base += off;
      printf("Prog block %d (%d-%d)\n", block, off, off + size);
      memcpy(bd->data + base, buffer, size);
      bd->cycles[block] ++;
      return 0;
  }
  
  static int erase(const struct lfs_config* c, lfs_block_t block) {
      struct mockbd* bd = (struct mockbd*)c->context;
      //Check range
      assert(block >= 0 && block < bd->blkcnt);
  
      //Check requirement of erased
      bool blockempty = true;
      int base = block * bd->blksize;
      for (int i = 0; i < bd->blksize; i++) {
          if (!bd->cleaned[base + i]) {
              blockempty = false;
          }
      }
  
      if (blockempty) {
          printf("Duplicated erase blk %d\n", block);
          return 0;
      }
  
      printf("Erase block %d\n", block);
      for (int i = 0; i < bd->blksize; i++) {
          bd->cleaned[i + base] = true;
          bd->proged[i + base] = false;
          bd->data[i + base] = 0;
      }
      bd->cycles[block] ++;
      return 0;
  }
  
  static int sync(const struct lfs_config* c) {
      struct mockbd* bd = (struct mockbd*)c->context;
      return 0;
  }
  
  void feedcfg(const mockbd* bd, struct lfs_config* cfg) {
      cfg->context = (void*)bd;
      cfg->read = read;
      cfg->prog = prog;
      cfg->erase = erase;
      cfg->sync = sync;
  
  
      cfg->read_size = bd->blksize;
      cfg->prog_size = bd->blksize;
      cfg->block_count = bd->blkcnt;
      cfg->block_size = bd->blksize;
      cfg->cache_size = bd->blksize;
      cfg->block_cycles = 32;
      cfg->lookahead_size = 128;
  }
  
  static const char CYCLECHARS[] = "0123456789abcdefghijklmnopqrstuvwxyzABCDEFGHIJKLMNOPQRSTUVWXYZ~!@#$%^&*()_+-=[]{};':";
  void dumpcycles(const mockbd* bd) {
      printf("Cycles list:");
      for (int i = 0; i < bd->blkcnt; i++) {
          putchar(CYCLECHARS[bd->cycles[i] % (sizeof(CYCLECHARS) - 1)]);
      }
      putchar('\n');
  }
  
  #define LFS_TEST(x) { \
	  int err = x; \
	  if (err < 0) { \
		  printf("Err at func %s, line %d, err %d\n", __func__, __LINE__, err); \
		  return 0; \
	  } \
  }
  
  extern "C" int lfs_fs_rawtraverse(lfs_t * lfs,
	  int (*cb)(void* data, lfs_block_t block), void* data,
	  bool includeorphans);
  
  int blkdump(void* data, lfs_block_t block) {
	  printf(" %d", block);
	  return 0;
  }
  int main()
  {
	  const int blksize = 4096;
	  mockbd bd;
	  struct lfs_config cfg = {};
	  unsigned char* ordered_buf = (unsigned char*)malloc(blksize);
	  assert(ordered_buf);
	  unsigned char* read_buf = (unsigned char*)malloc(blksize);
	  assert(read_buf);
  
	  makebd(&bd, 1024, 400);
	  feedcfg(&bd, &cfg);
	  lfs_t fs;
	  LFS_TEST(lfs_format(&fs, &cfg));
  
	  LFS_TEST(lfs_mount(&fs, &cfg));
  
	  const char* fname = "test_3.txt";
	  {
		  unsigned short* shortbuf = (unsigned short*)ordered_buf;
		  for (int j = 0; j < 3774 / 2; j++)
			  shortbuf[j] = j + 3;
	  }
	  lfs_file_t file;
  
	  //Make a 3774 size file
	  LFS_TEST(lfs_file_open(&fs, &file, fname, LFS_O_RDWR | LFS_O_CREAT));
	  LFS_TEST(lfs_file_write(&fs, &file, ordered_buf, 3774));
	  LFS_TEST(lfs_file_close(&fs, &file));
  
	  //Dump blocks
	  printf("Blocks in use:");
	  lfs_fs_rawtraverse(&fs, blkdump, NULL, true);
	  printf("\n");
  
	  //Truncate to 1024
	  LFS_TEST(lfs_file_open(&fs, &file, fname, LFS_O_RDWR));
	  LFS_TEST(lfs_file_truncate(&fs, &file, 1024));
	  LFS_TEST(lfs_file_close(&fs, &file));
  
	  //Dump blocks
	  printf("Blocks in use:");
	  lfs_fs_rawtraverse(&fs, blkdump, NULL, true);
	  printf("\n");
  
	  return 0;
  }
```

Result I got:

```
Read unknown data blk 1, off 0, size 1024
Erase block 0
Prog block 0 (0-1024)
Erase block 1
Prog block 1 (0-1024)
Erase block 0
Prog block 0 (0-1024)
Erase block 304
Prog block 304 (0-1024)
Erase block 305
Prog block 305 (0-1024)
Erase block 306
Prog block 306 (0-1024)
Erase block 307
Prog block 307 (0-1024)
Erase block 1
Prog block 1 (0-1024)
Blocks in use: 0 1 307 306 305 304
Erase block 0
Prog block 0 (0-1024)
Blocks in use: 0 1 305
```

Result I expect:

```
Read unknown data blk 1, off 0, size 1024
Erase block 0
Prog block 0 (0-1024)
Erase block 1
Prog block 1 (0-1024)
Erase block 0
Prog block 0 (0-1024)
Erase block 304
Prog block 304 (0-1024)
Erase block 305
Prog block 305 (0-1024)
Erase block 306
Prog block 306 (0-1024)
Erase block 307
Prog block 307 (0-1024)
Erase block 1
Prog block 1 (0-1024)
Blocks in use: 0 1 307 306 305 304
Erase block 0
Prog block 0 (0-1024)
Blocks in use: 0 1 305 304
```

rvanschoren, 2022-12-22:
Apologies for the verbal feedback rather than nice test code, but I think I ran into the same issue. I'm still using LittleFS1.3, but bug is still present in master branch as well.

Problem is happening when you truncate to a smaller size that is on a block boundary. I'm using 4096 block size, so file sizes that correspond to full block usage are: 4096, 8188, 12276, 16368, 20452,...

Issue happens when you create a file, e.g. size 4100 bytes. This will result in 2 blocks. If you then seek back to position 4096 and truncate then the head will incorrectly point to the 2nd block.

Reading from this file will then return bad data.

Additionally, if you do a lfs_ctz_traverse, this file will think that the head (still pointing to 2nd block) is actually the first block because lfs_ctz_index returns 0 for size 4096. This will start to cause issues when the filesize is 20452 because then the traverse function will assume there are 2 previous pointers in blocks with one previous pointer (and vice versa). This will result in reading blocks that don't exist.

I think the solution is to fix the head properly in lfs_file_truncate (lfs.c:1774) by using "size > 0 ? size - 1 : 0" instead of "size" as parameter to lfs_ctz_find.

[image]

Hope this helps you.

colin-foster-in-advantage, 2023-01-26:
I created #766 as a verification. Fails the check, as expected.

colin-foster-in-advantage, 2023-01-26:
I can also confirm that the fix from @rvanschoren makes the test I submitted above succeed.

geky, 2023-04-17:
Hi all, thanks for creating an issue for this. Sorry that it took me so long to address this issue.

I've built on @colin-foster-in-advantage's test case with a fix here: https://github.com/littlefs-project/littlefs/pull/800

@rvanschoren's solution is correct, though I've tweaked it a bit since the special case where `size == 0` should really result in the file reverting to an "inline file".

Let me know if anything looks wrong there.

PR #800 (2023-04-26): Fix block-boundary truncate issues
Building on the work from @colin-foster-in-advantage, @rvanschoren, and @hgspbs, I believe this fixes #732, #268, and likely others.

Long story short, `lfs_file_truncate` has had a bug for a while where truncates to block boundaries (after adjusting for CTZ skip-list metadata) would corrupt the internal state of the CTZ skip-list.

This fix started with @colin-foster-in-advantage's test case that made it easy to reproduce the issue, but I've taken the idea and expanded the existing truncate tests to cover both this and other possible off-by-one issues.

A de [... cut]