littlefs-project/littlefs #733: Is `lfs_dir_read/seek/rewind` useable if there is changes to the content of the dir?

Xywzel, 2022-10-04:
So I have need to implement recursive delete of files and directories in path. It would be quite helpful if the functions that operate on directories told if the `lfs_dir_t` and `struct lfs_info` data from them are still useable if you make changes inside the directory. Say if I remove a file from directory, can I still continue iterating over the directory with `lfs_dir_read`? Do I need to close the directory before opening a child directory?

e107steved, 2022-10-05:
Seems to have worked fine for me, although I've not used the function heavily. My code calls it through the universal 'readdir()' function, so there may be a few other bits of code in the way

geky, 2022-10-24:
Yes. There is actually an explicit test for this, see [[here]](https://github.com/littlefs-project/littlefs/blob/40dba4a556e0d81dfbe64301a6aa4e18ceca896c/tests/test_dirs.toml#L587). And I know that some rather critical software depends on this.

It's a fascinating requirement from a design standpoint, a CS purist would argue it should be undefined behavior, but then how the heck are users supposed to delete stuff in directories.

This works because keeps track of all open `lfs_dir_t/lfs_file_t`s in a linked-list and updates their state when lfs_remove or other metadata changing operations are called.