littlefs-project/littlefs #741: ISO usage documentation

CCurl, 2022-10-26:
I may be wrong, but the info here seems to be mostly geared towards developers implementing littlefs for their boards, not for people using littlefs. Is that accurate?

My goal is to be a user of littlefs. I am trying to write my code so that it will work with any board that supports littlefs ... for example Teensy4 or RPI PICO, or ESP8266 boards. I shouldn't have to know or care what board I am using, or the specifics of the implementation.

To that end, I'm looking for a concrete example of how I would use it as an end user (not an implementer). For example, the very first call lfs_format() takes a (lfs_t *lfs), which is a non-trivial structure. as an end user, how am I to how to construct this lfs_t structure I need to pass?

int lfs_format(lfs_t *lfs, const struct lfs_config *config);

CCurl, 2022-10-26:
Essentially, I am trying to create a littlefs-based version of my own little wrapper for a few typical file operations that can be used on either the PC (Windows or Linux) or any other system that supports files ... an API with a feel like a subset of the POSIX file API:

// FILEs - the c4 FILE API
extern void fileInit();
extern int fOpen(char *fn, int md);
extern void fClose(int fh);
extern int fRead(char *buf, int sz, int fh);
extern int fWrite(char *buf, int sz, int fh);
extern int fGetC(int fh);
extern void fPutC(int c, int fh);
extern int fGetS(char *buf, int sz, int fh);
extern void fDelete(char *fn);
extern void fList();

e107steved, 2022-10-28:
I think you need to read the littleFS documentation a bit more.
Essentially littleFS itself is totally hardware-independent. A higher layer (outside littleFS) defines and implements the platform-specific bits. So you will need to implement hardware interface routines for each supported platform.
Your objective is perfectly achievable; I (and no doubt others here) have routines which integrate littleFS, FAT and other file systems into a unified system.
Have a look at the mbed project for inspiration and maybe some code you can use.

CCurl, 2022-11-01:
> I think you need to read the littleFS documentation a bit more. Essentially littleFS itself is totally hardware-independent. A higher layer (outside littleFS) defines and implements the [...]

Thanks for your reply. I agree ... I should read the documentation a little more, if I could find it :). I mention in the title in the title that I am ISO (in search of) usage documentation. Perhaps you could point me to where I could find the user documentation?

e107steved, 2022-11-02:
Yes, the "user-facing" interface definition and implementation is down to the platform-specific implementation - you could choose to just provide a platform specific initialisation routine, and leave the user to access the file system through the rest of the littlefs API (documented in lfs.h). Or you could provide a shim which uses a slightly different interface for consistency or other reasons. 

Compared with many open source projects, I've found the documentation pretty good - there's an overview in the main project page (https://github.com/littlefs-project/littlefs) which also points to some reference documents. Then the API detail in lfs.h (which is generally POSIX-like). There's also a lot of useful detail in some of geky's responses in these 'issues'; he's very good at explaining things in detail, including the reasons for some of the choices. This last bit could do with extracting and consolidating (and no, I'm not volunteering at present).

As an aside, to me, 'ISO' is a standards organisation - never come across it being used to mean 'in search of' before.

geky, 2022-11-04:
To be fair the littlefs documentation could be improved. It's something I will eventually get to but it's a bit difficult to know the balance of improving documentation vs progressing technical problems in littlefs.

Something to consider is the different ecosystems of PCs/Linux-scale systems and microcontroller-scale systems. In the microcontroller space hardware can be quite varied and is often application-specific. So depending on the complexity of the application it's common for code to be written "baremetal", talking to hardware directly and organizing itself with an event-loop/RTOS/other scheduler. In this case the definition of "operating-system" gets a bit eroded, which is probably the reason for the confusion around user vs implementer. They're often the same person in this space.

There have been several projects to standardize things and make microcontroller development look more like Linux-scale development: [Zephyr](https://www.zephyrproject.org/) and [Mbed OS](https://os.mbed.com/) to name a couple. But due to how diverse and historically-unstandardized the hardware is, it's a hard problem. I think it's easy to underestimate the effort those teams put in to make things behave remotely consistent.

Both Zephyr and Mbed OS and others provide littlefs in their storage stacks. That may be something to consider if you just want things to work across a bunch of devices. However they bring their own complexity.

---

Also something that you may run into: ESP8266 is a different architecture (Xtensa) from Teensy4/RPI PICO/ESP32 (Arm Cortex-M). Neither Zephyr and Mbed OS support it. I've seen an Arduino fork that supports the ESP8266 and contains LittleFS: https://github.com/esp8266/Arduino, but not used it.

closed 2023-07-01 by CCurl