littlefs-project/littlefs #748: Power loss after erase but before write cause error LFS_ERR_CORRUPT during mount

vome101, 2022-11-17:
I observe that the file system would become corrupt if there is a power loss during writing a file. I made this happen by putting a break point at the erase and then repower the device before it can write. On the next mount, the file system will return LFS_ERR_CORRUPT during mount. Is this a known issue and is there any method to fix this?

LadislavOzobot, 2022-11-22:
Hi! We are currently debugging issue that may be related to this. What's interesting is that it appears filesystem read operations still work ok-ish, but write never does. We're still investigating, but have an image that we can share (privately) that seems to reproduce this issue.

Any progress on this issue? @vome101 as you seem to be able to reproduce the problem reliably, can you perhaps prepare a minimal filesystem that exhibits it and does not contain any confidential data?

vome101, 2022-11-22:
I found the issue is caused by the flash translation layer. Whenever there is a page (256 bytes) write, the FTL would read out a block (4096 bytes) to memory, replace the page in memory, erase the block, and then write the block back. If I repower the device after the erase but before the write, it would corrupt the file system. The previous entries in the block are erased. I'm not sure why this would cause the file system to be corrupt.

I fixed the issue by checking the page to be written to see if it is all FF. If it is all FF, the FTL won't erase and write directly. If there is other data in the page, the block will be erased and then written.

LadislavOzobot, 2022-11-22:
Thanks for the feedback. I see this is not the problem we're facing. We'll continue investigating tomorrow. I guess this issue is OK to close, as the bug turns out not to be a problem of littlefs?

vome101, 2022-11-22:
My issue isn't a problem of littlefs. I will close the issue.

If write fails, it may be the page being written isn't erased (all 0xFF). I have some corrupted file system binaries and I see that there are pages that is all zeroes. It looks like the erase didn't erase the entire block.

closed 2022-11-22 by vome101