littlefs-project/littlefs #749: Discussion: SPI vs SDIO when using littlefs

klemdev, 2022-11-23:
Hi.
I would like to ask for an opinion.

Currently I am using littlefs for accessing SD card via SPI at 20MHz.
Configured with block size 512 and read/prog/cache sizes same 512.
I need to periodically append 512B of data to a file.

I saw some previous discussions regarding file write and close duration.
In my case, each take about 50ms.

I would like to ask if anyone has any experiences with SDIO compared to SPI when using littlefs?
I am tempted to write an SDIO driver for my STM32F4, but I doubt it would improve timeouts.

Why?
Because if my measurements are correct, the whole append of 512B including open(), seek(), write() and close() takes about 115ms, but of that only about 35ms is bare SPI communication. All other time is spent elsewhere.

Would really appreciate any comment on that. It is clear from previous discussions that write() and close() is hard to impossible to optimize/speed up so I am trying to find any other way to speed up my application.

Thank you very much for your comments.

geky, 2022-12-02:
Hi @klemdev,

I would normally comment that you should measure to see where your bottleneck is, but it sounds like you already did that and found it wasn't the IO bus.

Generally what I've seen is SDIO/QSPI/faster-bus can greatly speed up read-heavy applications, but they don't have as much an impact when your application needs to erase often. Erasing a block is very time-consuming, so Amdahl's law kicks in and you get diminishing returns from faster data transfer.

This gives me an excuse to share one of my favorite pictures from when I looked at something similar previously: https://github.com/ARMmbed/mbed-os/pull/4207

> As for why we don't see a proportional increase in throughput on the SD card, well I think this image sums it up quite nicely:
> [image]

---

Some filesystems improve this by allowing you to pre-erase blocks in the background, sometimes called/integrated into garbage collection. Unfortunately littlefs can't provide this because it currently doesn't have a mechanism for tracking which blocks have been erased. It might be an area of optimization in the future.

klemdev, 2023-03-23:
Hi @geky,

Sorry for delayed reply, but thank you very much for your comment.

closed 2023-03-23 by klemdev