littlefs-project/littlefs #750: Investigating fs corruption - LFS_ERR_CORRUPT due to corrupt skiplist

LadislavOzobot, 2022-11-25:
Hello!
We're currently investigating filesystem corruption on a device we got back from customer. Writing to the disk (or reading specific files) results in `LFS_ERR_CORRUPT`. We can easily reproduce it with binary image and littlefs-tools package:

```
$ littlefs_extract -b 4096 -c 4035 -i corrupted_56B9C1.img -d /tmp/x
CRITICAL - 121 - LittleFSError -84: LFS_ERR_CORRUPT
```

But the scripts/readtree.py works fine and does not report issues. This supports our following findings, as readtree.py does not parse skiplists.

Our investigation shows there is a problem in a skiplist that points out of range of the flash. This issue seems be hit for example when the allocator walks the flash.

We understand that skiplist is protected by CRC and if it's incorrect, the entry is disregarded and the file (or part of the data) is lost. However, in this specific case it causes the filesystem to be in unrecoverable state - the CRC seems to be correct.

We can share the filesystem image that reproduces the problem (16MB), but would prefer to do it privately. @geky, can you share your email address or reach out to be at laska@ozobot.com and help us investigate the issue? The investigation has two parts:

- What is possible cause of the corruption? This may be hard to judge from the sample. However, it'd be interesting to know if the issue looks like bad data intentionally written by littlefs (due to other bug, either in littlefs or the rest of the system) or corruption that is not properly detected.
- Why is the issue unrecoverable (this may be a design decision?) and possibly figure out how to recover (in our case, we can just drop the data, as it will be synchronized eventually).

LadislavOzobot, 2022-11-25:
Additional log showing the suspected bad skiplist is at 1153:
```
Go through directory pair[1898,1899] tail [1265, 1266] with 60 entries
		 CTZ head 1420 size 22474
			Block read status 0
			Block read status 0
			Block read status 0
		 Traverse error 0
		 CTZ head 1961 size 38372
			Block read status 0
			Block read status 0
			Block read status 0
			Block read status 0
			Block read status 0
		 Traverse error 0
		 CTZ head 1041 size 16320
			Block read status 0
			Block read status 0
		 Traverse error 0
		 CTZ head 2254 size 16776
			Block read status 0
			Block read status 0
		 Traverse error 0
		 CTZ head 1568 size 12860
			Block read status 0
			Block read status 0
		 Traverse error 0
		 CTZ head 1290 size 12892
			Block read status 0
			Block read status 0
		 Traverse error 0
		 CTZ head 1755 size 15348
			Block read status 0
			Block read status 0
		 Traverse error 0
		 CTZ head 2205 size 9632
			Block read status 0
		 Traverse error 0
		 CTZ head 1466 size 32557
			Block read status 0
			Block read status 0
			Block read status 0
			Block read status 0
		 Traverse error 0
		 CTZ head 1742 size 32925
			Block read status 0
			Block read status 0
			Block read status 0
			Block read status 0
		 Traverse error 0
		 CTZ head 2816 size 49234
			Block read status 0
			Block read status 0
			Block read status 0
			Block read status 0
			Block read status 0
			Block read status 0
		 Traverse error 0
		 CTZ head 1935 size 10540
			Block read status 0
		 Traverse error 0
		 CTZ head 2083 size 33032
			Block read status 0
			Block read status 0
			Block read status 0
			Block read status 0
		 Traverse error 0
		 CTZ head 2094 size 21564
			Block read status 0
			Block read status 0
			Block read status 0
		 Traverse error 0
		 CTZ head 1673 size 12836
			Block read status 0
			Block read status 0
		 Traverse error 0
		 CTZ head 1425 size 17424
			Block read status 0
			Block read status 0
		 Traverse error 0
		 CTZ head 1652 size 45292
			Block read status 0
			Block read status 0
			Block read status 0
			Block read status 0
			Block read status 0
			Block read status 0
		 Traverse error 0
		 CTZ head 2212 size 25416
			Block read status 0
			Block read status 0
			Block read status 0
		 Traverse error 0
		 CTZ head 1286 size 13484
			Block read status 0
			Block read status 0
		 Traverse error 0
		 CTZ head 2581 size 12924
			Block read status 0
			Block read status 0
		 Traverse error 0
		 CTZ head 2366 size 18082
			Block read status 0
			Block read status 0
		 Traverse error 0
		 CTZ head 1534 size 19360
			Block read status 0
			Block read status 0
		 Traverse error 0
		 CTZ head 2768 size 60785
			Block read status 0
			Block read status 0
			Block read status 0
			Block read status 0
			Block read status 0
			Block read status 0
			Block read status 0
		 Traverse error 0
		 CTZ head 1153 size 91443
			Block read status 0
			Block read status 0
			Block read status 0
			Block read status 0
			Block read status -84
		 Traverse error -84
```

zhu-andrew, 2024-03-29:
Hi @LadislavOzobot did you ever figure out the cause of the corruption? We're seeing a very similar issue where at some point `lfs_ctz_traverse` reads a new head that points outside the flash bounds resulting in an `LFS_ERR_CORRUPT`

LadislavOzobot, 2024-04-01:
Hi @zhu-andrew , no. We also did not have any other incident ever since, so we dropped further investigation.