littlefs-project/littlefs #751: I'm trying to use littles to create new file with fixing/mapping to physical address

votuananhs, 2022-11-29:
Hello all,

I have ported littlefs running with internal flash of STM32 MCU. I'm trying to use it to receive new application. But I have one issue when I can't fix physical address of this file to special area. Purpose of that, when the MCU's program jump into boot loader. It can read out data and rewrite it into main application area for FOTA. Does someone have same this issue with me before ? and could you please share method/solution to do that ?

alexandrehagihara, 2022-11-29:
Use another reserved region and copy the file content to this region after downloading.

votuananhs, 2022-11-29:
thank you @alexandrehagihara 

I have thought that solution. But that means, I should split memory into 3 partitions memory:
- Main Application
- Store new Application
- Sharing memory area between App and Bootloader.
With current my MCU it just has 128kbytes flash, so each partition will have around 42kbytes memory. It will limit size of output application. If I have no any other good solution. I should integrate littlefs into boot loader and split internal flash memory into two main partitions: Main Application and Store New Application.

sigsstock, 2022-11-30:
@votuananhs One thing you may want to think about with the internal memory layout for the STM32 is the sector and page alignments. The recent parts I've been working with had the following.

Smallest erasable size is 1 page (128 Bytes)
Smallest protectable size is 1 sector (4KB)

So when I was figuring out the memory layout I decided to keep each section on sector boundaries at the start of development so that we we determined sections needed to be write protect we could. Usually the bootloader will end up being smaller then the application depending on what it needs to do which was the case for me so I ended up with the following layout. 

<html>
<body>

Start Address | End Address | Size (KB) | Purpose
-- | -- | -- | --
0x0800 0000 | 0x0800 5FFF | 24 | Bootloader
0x0800 6000 | 0x0801 2FFF | 52 | Main Application
0x0801 3000 | 0x0801 FFFF | 52 | Fallback Application

</body>
</html>

I did not use littlefs for the bootloader in this project though so I can't speak to that. Hope this gives you some ideas as you work on yours though.

e107steved, 2022-12-02:
@votuananhs - my unserstanding of the way littleFS works is that what you want to achieve simply isn't possible. Each time a file is updated, littleFS rewrites parts of the file, including the changed data - always in a previously empty block, and then creates an updated directory entry, again in a free block.

So I think that you will either have to adopt  @alexandrehagihara's suggestion, or incorporate littleFS in your bootloader.

votuananhs, 2023-11-13:
Please close this ticket

closed 2023-11-14 by geky