littlefs-project/littlefs #755: LFS_ERR_NOSPC returned by lfs_dir_compact() when there is still room on the disk

ajheck, 2022-12-06:
Hello all,

I am running into an issue where littlefs fails to close a file once a certain number of files are present in the root directory.  Specifically, the check 

`if (commit->off + dsize > commit->end) {
        return LFS_ERR_NOSPC;
    }`

in lfs_dir_commitattr will return LFS_ERR_NOSPC with the following call-stack:

lfs_file_close ->
lfs_file_rawclose ->
lfs_file_rawsync ->
lfs_dir_commit ->
lfs_dir_orphaningcommit ->
lfs_dir_relocatingcommit ->
lfs_dir_splittingcompact ->
lfs_dir_compact ->
lfs_dir_traverse ->
lfs_dir_commit_commit ->
lfs_dir_commitattr

The littlefs configuration is typically as follows:

  read_size = 4096
  prog_size = 4096
  block_size = 262144
  block_count = 2032
  block_cycles = 500
  cache_size = 4096
  lookahead_size = 64
  name_max = 255
  file_max = default
  attr_max = 1022
  metadata_max = 4096

When this error occurs, littlefs is using less than 1% of the total space allocated for it (508 MB).  The number of files required to cause this issue changes with the configured metadata_max. If metadata_max is 4 kB, the issue happens when closing the 124th file.  If metadata_max is 512 B, the issue happens when closing the 15th file.  It is my understanding that littlefs should move metadata around so that many files can be stored in a directory, and that metadata_max should not have an effect on the total number of files which can be placed in a directory.  Are these assumptions correct?

I've taken a look at https://github.com/littlefs-project/littlefs/issues/478, and the files only have a small custom attribute containing an application level checksum.  Additionally, the read_size and prog_size are not the same size as block_size, which was another suggested fix in the linked ticket.

dpkristensen, 2022-12-06:
I'm having this exact same issue with basically the same configuration.  One of the things that seems to be happening is that metadata_max is less than the block size, which seems to cause it to artificially run out of space.

The backend here is directly to an unmanaged NAND, so changing the block size is not possible.  Also in this case I see the performance of lfs_file_opencfg() gradually degrades as the number of files.  Increasing the size of metadata_max will increase the # of files I can open, but also severely degrades lfs_file_opencfg() performance.

I know @geky has previously mentioned using a B-tree instead of a CTZ list; I think this is possibly the only thing that will improve performance in this situation.

pedromarinho94, 2024-07-29:
I'm having the same issue.

geky, 2024-10-04:
Hi all, sorry for the late response.

I've been digging around to try to figure out these LFS_ERR_NOSPC errors and found one bug that may be related: https://github.com/littlefs-project/littlefs/pull/1031

It would be interesting to know if this fixed @ajheck's issue, since it looks like you're using the same `prog_size` and `metadata_max`.

I'm not aware of an issue when `metadata_max` > `prog_size`, though it's certainly possible and would be good to know about.

geky, 2024-10-04:
Just some other -- late -- comments:

> It is my understanding that littlefs should move metadata around so that many files can be stored in a directory, and that metadata_max should not have an effect on the total number of files which can be placed in a directory. Are these assumptions correct?

Yes, this is how it's supposed to work normally. Directories are linked-lists of metadata blocks, and `metadata_max` just limits how much metadata is stored in each block. If you have many files in a directory -- and no fs bugs -- these can end up in multiple metadata blocks.

The one catch is that a single file's metadata can't span multiple metadata blocks. So if you have, say, a large filename or many custom attributes, it can cause problems.

> I know @geky has previously mentioned using a B-tree instead of a CTZ list; I think this is possibly the only thing that will improve performance in this situation.

I don't think this specific feature will help with `lfs_file_opencfg` speed, since directories in littlefs are not stored in files, but I've been looking at a few other ideas such as red-black trees in logs.