littlefs-project/littlefs #767: can littlefs works on iic eeprom chip, such as 24C256?

ZanoZ, 2023-01-30:
Dear my friend:
Can littlefs work with eeprom through iic interface? I tried but still have some err, I am not sure if it can work with eeprom chip such as 24c256 or somekind chips.
If it is yes , what's the points that we should carefully do with? 
Thanks for your time  reading this ,and appreciate for any words in reply!

BR.
Zon

ZanoZ, 2023-01-30:
well, I found topic https://github.com/littlefs-project/littlefs/issues/449
it is a looming lighthouse， leading me ...
I will post sth follow this to make it clear if I succeed to touch it.

ZanoZ, 2023-01-30:
well, at last I got these message:
..\littlefs\lfs.c:5339:trace: lfs_mount(20004e48, 080285a0 {.context=00000000, .read=08008131, .prog=08008109, .erase=08008103, .sync=08008159, .read_size=1, .prog_size=1, .block_size=128, .block_count=128, .block_cycles=500, .cache_size=16, .lookahead_size=16, .read_buffer=20004b48, .prog_buffer=20004c48, .lookahead_buffer=20004d48, .name_max=0, .file_max=0, .attr_max=0})
..\littlefs\lfs.c:1224:error: Corrupted dir pair at {0x0, 0x1}.
..\littlefs\lfs.c:5358:trace: lfs_mount -> -84
..\littlefs\lfs.c:5308:trace: lfs_format(20004e48, 080285a0 {.context=00000000, .read=08008131, .prog=08008109, .erase=08008103, .sync=08008159, .read_size=1, .prog_size=1, .block_size=128, .block_count=128, .block_cycles=500, .cache_size=16, .lookahead_size=16, .read_buffer=20004b48, .prog_buffer=20004c48, .lookahead_buffer=20004d48, .name_max=0, .file_max=0, .attr_max=0})

I use the setting try to run the littlefs on a 24C512(512kbit/64kByte):
const struct lfs_config cfg =
{
	// block device operations
	.read  = lfs_deskio_read,
	.prog  = lfs_deskio_prog,
	.erase = lfs_deskio_erase,
	.sync  = lfs_deskio_sync,

	// block device configuration
	.read_size = 1,
	.prog_size = 1,
	.block_size = 128,
	.block_count = 128,	//
	.cache_size = 16,
	.lookahead_size = 16,
	.block_cycles = 500,

	//
	//
	.read_buffer = read_buffer,
	.prog_buffer = prog_buffer,
	.lookahead_buffer = lookahead_buffer,
};

any commit will be helpful~ Thanks ~ :''(

aentinger, 2023-03-23:
Yes it can, I just tried it for myself, see [here](https://github.com/107-systems/107-Arduino-littlefs/blob/main/examples/EEPROM/EEPROM.ino).