littlefs-project/littlefs #773: Assistance in implementing littleFS on TI C2000 with SYS/BIOS?

ken-at-em, 2023-02-02:
Hi There,

Apologies if this is not the appropriate channel. If there is a better way to have this discussion please advise.

We are looking for some freelance/consulting support to get littlefs up and running on our embedded target. It's a little tricky as a "byte" on our target is 16bits instead of 8bits, we're trying to work around it but are a bit resource strapped.

Any help at all is appreciated. Willing to pay competitive freelance rates for advice, consulting or freelance implementation.

Thanks,
Kenneth Miller
Erdos Miller
ken@erdosmiller.com

sicrisembay, 2023-03-28:
TI C2000 MCUs have its quirks where sizeof(byte) == sizeof(short).
On my free time, I've done some port of XCP (address granularity of WORD) and CANopenNode to C2000 (SM320F28335-HT) with SysBIOS (6.83) and source code are publicly available in github.  I'm currently looking into porting littleFS to C2000.  I think that this is very feasible.

I think that we are working on the same industry.  So, I can't do freelance.  I'm open to discussions as long as it is done in open discussion board/forum.

cheers
Sicris

sicrisembay, 2023-04-15:
After spending two weekends, I'm able to run LFS on my C2000 microcontroller.  This can be found in [littlefs port](https://github.com/sicrisembay/littlefs), branch: c2000_port.  All code changes are wrapped in  #if LFS_TI_C2000.  This is  a quite messy workaround and I can't think of a more elegant approach.

I've also added an [example project](https://github.com/sicrisembay/littlefs_c2000).  In this project,  I'm using TMS320F28335 with SPI FRAM (FM25W256).

I've only tested some basic functions (i.e. format, mount, creating file, write to file, read to file, etc).  This is not extensively tested.  I'm hoping that someone in the opn-source community will take interest and can help do some test.

@ken-at-em, if you have time, please take a look of the port.  Please play around with it and let me know for any issues with this port.

cheers
Sicris

vonj, 2023-04-15:
aerdos/littlefs_c2800: LittleFS port to the C2800 architecturegithub.comThere is also this one15 apr.

sicrisembay, 2023-04-16:
@vonj Thanks for sharing another c2800 port project.  Link to project pointed by vonj ([aerdos/littlefs](https://github.com/aerdos/littlefs)).   I've checked out the said c2800 port.  It looks like much porting effort is still needed as many LittleFS featues are not working on this port.

sicrisembay, 2023-04-29:
Got some weekend spare time.  I've added YMODEM to [littlefs_c2000](https://github.com/sicrisembay/littlefs_c2000).  This is to test file transfers between MCU and host PC... also, a good addtional test of LittleFS C2000 port.