littlefs-project/littlefs #777: lfs_rename overwrites an existing file without any warning

lf-novelt, 2023-02-22:
Hello

I just discovered this behavior. I was expecting `lfs_rename`  to return an error if the destination already exists.

But instead, it simply overwrites the destination and gladly returns a success code (aka 0 aka LFS_ERR_OK) .

In the header, it seems like the expected behavior but most of the file systems instead return an error when the target already exists
https://github.com/littlefs-project/littlefs/blob/6a53d76e90af33f0656333c1db09bd337fa75d23/lfs.h#L463-L469

Shoudn't littlefs do the same? ie not overwrite by default and return something like LFS_ERR_EXIST aka Entry already exists instead ?

hattesen, 2023-03-06:
Although the behavior of `lfs_rename` is in line with [POSIX `rename`](https://pubs.opengroup.org/onlinepubs/000095399/functions/rename.html), `lfs_rename` does not require write permissions to the files referenced by `*oldpath` and `*newpath` (as such permissions are not part of LittleFS, except when operating on an opened file – through lfs_open_flags). Also, a workaround like the one you proposed has a substantial runtime cost. So I find it reasonable to request LittleFS to support a rename mode that returns an error if `*newpath` already exists.

To maintain backward compatibiility though, the existing (default) behavior of `lfs_rename` should not be changed. But the `lfs_rename` function could be overloaded with an additional (optional) parameter, e.g. `boolean silentOverwrite = true`. This new parameter will control the behavior when renaming to an existing pathname, with a default value that maintains the current behavior, to one of:

1. Silently remove any existing file or (empty) directory referenced by `*newpath` (current behavior).
2. Return error `LFS_ERR_EXIST` if `*newpath` already exists (new, optional behavior).

lf-novelt, 2023-03-06:
thanks for your reply @hattesen, I understand that changing the default behavior of a function is quite tricky now.

Also I checked that adding optional parameters to a C function is not straighforward: https://stackoverflow.com/questions/1472138/c-default-arguments

So maybe we should do nothing except having this ticket might help to warn future users of this behavior, especially if they migration from Azure RTOS [fx_file_rename function](https://learn.microsoft.com/en-us/azure/rtos/filex/chapter4#fx_file_rename) 
[image]

hattesen, 2023-03-06:
@lf-novelt yes, you are quite right. I am generally coding C++, and tend to ignore the C restrictions ;)

However, something like this world still be possible, and mintain full backwards compatibility:

```C
// existing rename function signature with rename-overwrites-existing behavior
int lfs_rename(lfs_t *lfs, const char *oldpath, const char *newpath) {
    return lfs_rename(lfs, oldpath, newpath, true);
}

// new rename signature with parameterized rename-overwrites-existing behavior.
// Note: overwrite parameter may be implemented as enum for source code clarity and to allow more than two behaviors
int lfs_rename(lfs_t *lfs, const char *oldpath, const char *newpath, const bool overwrite);
```

lf-novelt, 2023-03-06:
yeah, still a C++ feature, C does not support multiple functions with the same name but different signatures  :P 
[image]

hattesen, 2023-03-06:
@lf-novelt aaaargh!

So the extended function will have to be renamed to maintai8n full compatibility, e.g.

````
// existing rename function signature with rename-overwrites-existing behavior
int lfs_rename(lfs_t *lfs, const char *oldpath, const char *newpath) {
    return lfs_rename_ex(lfs, oldpath, newpath, true);
}

// new rename signature with parameterized rename-overwrites-existing behavior.
// Note: overwrite parameter may be implemented as enum for source code clarity and to allow more than two behaviors
int lfs_rename_ex(lfs_t *lfs, const char *oldpath, const char *newpath, const bool overwrite);
```

geky, 2023-03-24:
Hi @lf-novelt , @hattesen , thanks for creating an issue.

As @hattesen points out the current behavior is in-line with POSIX's definition of `rename`:
https://pubs.opengroup.org/onlinepubs/000095399/functions/rename.html

It actually has a purpose as well, since `rename` is historically the only way to replace a file atomically in a multi-process system.

It's interesting to note there's a precedent for additional rename behaviors in Linux with the addition of `renameat2`. I believe RENAME_NOREPLACE provides the proposed behavior:
https://man7.org/linux/man-pages/man2/renameat.2.html

> I will gladly implement this change if approved

I say go for it, though a word of warning and apology if it takes me a long time to merge. My attention/bandwidth is currently focused on other larger-scale issues in the filesystem.

---

Personally, I would name it `lfs_renameexcl` to match the `LFS_O_EXCL` which provides the same behavior when creating files. I also don't know if you need the flag, since users could still call `lfs_rename` when needed.

Internally these could both be routed to the same `lfs_rawrename` function to deduplicate/reduce code size. The cost of more functions is mostly cognitive:

``` c
// existing rename function signature with rename-overwrites-existing behavior
int lfs_rename(lfs_t *lfs, const char *oldpath, const char *newpath) {
    // internal rename function
    return lfs_rawrename(lfs, oldpath, newpath, false);
}

// new rename signature with parameterized rename-overwrites-existing behavior.
int lfs_renameexcl(lfs_t *lfs, const char *oldpath, const char *newpath) {
    // internal rename function
    return lfs_rawrename(lfs, oldpath, newpath, true);
}
```

Having separate functions also increases the chance of code being gced at compile-time, though that may not provide much benefit here.

closed 2023-03-29 by lf-novelt