littlefs-project/littlefs #778: Our NAND part has HUGE blocks.. 256K. This means a 1K file wastes 99.6% of the block.

zammer2023, 2023-02-23:
Has anyone worked on improvements to increase utilization using pages within blocks? The part we are using has 2048 blocks containing 64 pages with 4096 bytes of storage. littleFS would be ideal if it could leverage pages and blocks, particularly since the page/block architecture is fundamental to NAND devices.

thanhminhmr, 2023-02-28:
Maybe a duplication of this issue/discussion: https://github.com/littlefs-project/littlefs/issues/11

geky, 2023-08-16:
Hi @zammer2023, thanks for creating an issue, the feedback is always useful.

I realize I'm late in responding, but I just wanted to note that this is going into some ongoing work to improve the file data-structure.

The planned improvement is to detach the RAM requirement from inline files, allowing inline files to be quite large without an unreasonable RAM requirement.

Inline files will still have a 4x storage overhead, something I'm not sure can be avoided, but this would at least be an improvement over the current 256x overhead you are seeing.