littlefs-project/littlefs #789: GCC Warning - 'strncpy' output truncated before terminating nul copying

baldhead69, 2023-03-28:
Hi,

Just for information, if littlefs project team want to fix the code.

```
../src/config/default/system/fs/src/sys_fs_littlefs_interface.c: In function 'LITTLEFS_stat':
../src/config/default/system/fs/src/sys_fs_littlefs_interface.c:411:13: warning: 'strncpy' output truncated before terminating nul copying as many bytes from a string as its length [-Wstringop-truncation]
             strncpy (stat->lfname, info.name, fileLen);
             ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
../src/config/default/system/fs/src/sys_fs_littlefs_interface.c:400:15: note: length computed here
     fileLen = strlen (info.name);
               ^~~~~~~~~~~~~~~~~~
```

geky, 2023-03-29:
Hi @baldhead69, thanks for raising an issue.

This actually looks like it's in a higher-level wrapper than the codebase here. Are you using a framework such as Mbed/Zephyr/Arduino?

baldhead69, 2023-03-29:
> Hi @baldhead69, thanks for raising an issue.
> 
> This actually looks like it's in a higher-level wrapper than the codebase here. Are you using a framework such as Mbed/Zephyr/Arduino?

Hi @geky ,

I am using microchip mplab code configurator( MCC ) to generate the code  for a pic32mz chip.

closed 2024-01-11 by baldhead69