littlefs-project/littlefs #793: Do I have to worry about page boundaries within my prog() function?

bselc, 2023-04-04:
Hi, I want to use littlefs with an SST26VF016B 16Mbit flash memory. The smallest block unit that can be erased is 4096 bytes. Within those 4096 bytes, there are 16 pages of size 256 bytes. Only one page can be programmed in one call. However, following the reference sheet, when more than 256 bytes is given to program, it wraps around the page start.

So, now I need to implement the prog() prototype of littlefs and if I undertand it correctly, prog() does only write x bytes within _one_ programmable block (in my case one page). I can calculate the memory address by the given argument block (number) and offset. However, what happens if given size exceeds the remaining number of bytes inside this page?

For example: 
```
block size = 4096, prog size = 256
prog(cfg, block=4, off=200, size=64)
```
would exceed a page boundary since the remaining number of bytes after the offset is smaller than to-write size=64.

I hope my question is understandable. I am kind of new to embedded / flash memory and hope to find an answer here.

e107steved, 2023-04-04:
I think you should be OK, certainly if you also have cache_size = 256 - I've been using another flash device in the same series for some time without noticing any problems, also with prog_size = 256 and block_size = 4096.
With cache_size = 256, littleFS maintains a write buffer of size prog_size, which logically must start on a 256-byte boundary otherwise there will be complications when wrapping into another block.
If cache_size is 512, say, I think the same argument applies.

Why not put a trap in your flash writing driver to check that writes don't cause a wrap around? That will settle it, unless @geky can give a definitive answer.

evpopov, 2023-04-05:
@bselc just a side note:
The datasheet -> section 5.2 second paragraph states: "The programmed data must be between 1 to 256 bytes and in whole byte increments"
To me that means you can do: .prog_size = .read_size = 1
At least that's what I'm doing with a similar chip and a page of 256 bytes, erase block of 4k and cache of 4k

As far as writing more than 1 page....  here's my quick and dirty code:

```
...
prvLfsCfg.prog  = &LFS_Write;
...

int LFS_Write(const struct lfs_config *c, lfs_block_t block, lfs_off_t off, uint8_t *buffer, lfs_size_t size)
{
	int32_t result = LFS_ERR_OK;
	uint32_t uiFlashAddr = (block * c->block_size) + off + SERIAL_FLASH_FS_START_ADDRESS;
	QSPI_SF_Write( uiFlashAddr, size, buffer);
	return result;
}

int32_t QSPI_SF_Write( uint32_t SF_Address, uint32_t Len, uint8_t *pSrcBuffer)
{
	xSemaphoreTake( prvQSPIMutex, portMAX_DELAY);
	// ensure we do not cross page boundaries
	uint32_t uiAddr = SF_Address;
	while (Len)
	{
		uint32_t uiPageOffset = uiAddr % prvSFlashPageSize;
		uint32_t uiWriteLen = min( Len, (prvSFlashPageSize - uiPageOffset));
		s25fl1xx_write(&sQSPI_Drv, pSrcBuffer, uiWriteLen, uiAddr, false);
		Len -= uiWriteLen;
		pSrcBuffer += uiWriteLen;
		uiAddr += uiWriteLen;
	}
	xSemaphoreGive(prvQSPIMutex);
	return 0;
}
```
Hope that helps.

Sorry got to go now so no time to format the code better, but I can try and explain more tomorrow if you can't read it.

geky, 2023-04-06:
Just to confirm, littlefs only promises to write in multiples of the prog_size. So for a 256 byte prog_size, littlefs may try to write 256, 512, 768, etc. What @evpopov provided is the correct way to batch this into page-size units for the flash device.

The prog_size config exists for devices that have a physical limit on the minimum number of bytes they can write at a time. Since the SST26VF016B supports byte-level reads/progs you can set both read_size/prog_size to 1.

---

In _theory_, you could set prog_size = page-size so that all program operations maximize the amount of data they send in a bus transaction. But I have never seen this be a bottleneck and the savings would probably be really small vs upgrading to QSPI/etc.

geky, 2023-04-11:
Ah yes, any read/prog operations will be multiples of the read_size/prog_size and aligned to the read_size/prog_size.

Assuming "start address" here is the `off` parameter in the read/prog functions.