littlefs-project/littlefs #794: Program stuck in "LFS_ASSERT(err <= 0);"

baldhead69, 2023-04-05:
Hi,

i am trying to run littlefs filesystem with FreeRtos in a pic32mz chip.

I am debugging the code and i verify that program not return from this assert "LFS_ASSERT(err <= 0);".
err = 1.

Function on lfs.c file:
```
static int lfs_bd_erase(lfs_t *lfs, lfs_block_t block)
{
    LFS_ASSERT(block < lfs->cfg->block_count);
    int err = lfs->cfg->erase(lfs->cfg, block);
    LFS_ASSERT(err <= 0);
    return err;
}
```

Some suggestion on what can cause this "error" ?

Notes:
PIC32MZ2048EFM144 chip.
DM320007-C development board.
SST26 sqi flash memory.

Thank's.

baldhead69, 2023-04-05:
Some suggestion ?

Maybe media not attached / working ?

alexandrehagihara, 2023-04-05:
The "erase" function is something platform dependent. You must implement it in a way that returns something valid as a lfs_error (enum defined in lfs.h). Your implementation is returning "1", a positive value.

My suggestion is that you modify it to return one of these 3:
    LFS_ERR_OK          = 0,    // No error
    LFS_ERR_IO          = -5,   // Error during device operation
    LFS_ERR_CORRUPT     = -84,  // Corrupted

baldhead69, 2023-04-05:
Hi @alexandrehagihara ,

I am using microchip mplab code configurator( MCC ) to generate the code for a pic32mz chip.

https://github.com/MicrochipTech/littlefs

https://github.com/littlefs-project/littlefs/issues/789

geky, 2023-04-06:
@alexandrehagihara is correct that this is a mistake in a level lower than littlefs. If the prog function returns "bytes written", there's a risk it expects littlefs to handle partial writes, which littlefs does not.

On the bright side, the reason this assert was added was to catch this exact mistake. So this is a good sign it's working (though it could have a comment).

Thanks @baldhead69 for opening an issue, but this is the wrong place for both this and #789. This project isn't owned or managed by Microchip, and we don't know much about what MCC is doing.

You can either contact Microchip's support, MCC's documentation may indicate where to go, or try to modify the generated code to work with littlefs.

baldhead69, 2023-04-07:
Thank's @geky

I understand.

I would just like suggestions on what could be causing this type of problem.

Media(sst26 flash memory in this case) not attached, or without fisical commmunication maybe.

What else could cause this kind of problem ?  

Any suggestion ?

Thank's.

geky, 2023-04-07:
Without more information we don't know, littlefs doesn't expect a positive number here. We just know the layer below isn't implemented to work with littlefs in this situation.

closed 2024-01-11 by baldhead69