littlefs-project/littlefs #795: Speed problem with external SPI Flash - W25Q64

gokul234sys, 2023-04-06:
When I write 1KB data for each file and then I to read it, it works perfectly.
But when I read 1KB of data and try to read it, the entire page to read time consuming is approx 30 sec
Anyone knows what might be the problem?
I use normal spi and my lfs config:
Config.read = blockDeviceRead;
	Config.prog = blockDeviceProg;
	Config.erase = blockDeviceErase;
	Config.sync = blockDeviceSync;
	Config.lock = lock;
	Config.unlock = unlock;
	Config.read_size = 256;
	Config.prog_size = 256;
	Config.block_size = 4096;
	// cfg.block_count = 1792; //Utilising only 7 MB
	Config.block_count =
		2048; // 1280(80 *16)5MB // 1024; // (4MB) 64*16 (Block count * Sector count)
	Config.lookahead_size = 16;
	Config.cache_size = 512;
	// Config.block_cycles = -1;
	Config.read_buffer = lfsReadBuf;
	Config.prog_buffer = lfsProgBuf;
	Config.lookahead_buffer = lfsLookaheadBuf;
	Config.block_cycles = -1;

geky, 2023-04-06:
Hi @gokul234sys, thanks for creating an issue.

I don't think there's enough info here to understand what's happening. There are several corner cases in littlefs with poor performance, but I'm not sure any are related to the size of the data you read.

Is it possible to reproduce this locally (off device)?

gokul234sys, 2023-04-24:
After erasing the device and reprogramming it, the writing speed is fast. However, once it writes 1MB of data, the speed becomes slow.

geky, 2023-04-26:
It's hard to know, but you may be hitting one of littlefs's performance issues: https://github.com/littlefs-project/littlefs/issues/797

admin456dnd, 2023-05-10:
Partition has 2 directories, each with 20 files and 200KB.
Writing 1MB
10KB to 20KB file size = 80-90mins
100KB to 400KB file size= 50 - 60 min
1MB file size = ~35mins
I already read #75 and i increased lookaheadsize but no improvement in speed ?
Any other suggestions ?